MarlinFirmware/Marlin #24890: [BUG] sd card does not start and lcd freezes

lipewww, 2022-10-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

A few months ago I decided to put Marlin on my Anet A6 printer, and since then I've never been able to use my sd card on the printer, I've had marlin since version 2.0.9.3, and since then it's never been fixed. When trying to start the card, the LCD freezes for about 10 seconds, then comes back as if nothing happened.

### Version of Marlin Firmware

2.0 or 2.1

### Printer model

Anet A6

### Bed Leveling

No Bed Leveling

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I formatted my computer and lost version 2.1 of the firmware that had some changes, but it didn't work anyway, I'll leave 2.0 available if you want to check it too
[Marlin-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/9800895/Marlin-2.0.x.zip)

ellensp, 2022-10-17:
Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Then provide the Configs for this version.

lipewww, 2022-10-17:
> Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If [...]

trust me, it's not resolved, I just lost the file to attach, it's on my printer.

lipewww, 2022-10-17:
> Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If [...]

aqui está uma configuração 2.1 bugfix
[config 2.1 bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/9801175/config.2.1.bugfix.zip)

ellensp, 2022-10-17:
bugfix is 02010200   this is several versions older at 02010000

ellensp, 2022-10-17:
Anet 10 has no sd detect pin. 
Will only know about sdcard when board is powered up with sd card in it, or on M21

lipewww, 2022-10-17:
I just tested in the new version, nothing changes. here is the settings
[config 2.1.zip](https://github.com/MarlinFirmware/Marlin/files/9801987/config.2.1.zip)

lipewww, 2022-10-17:
> Anet 10 has no sd detect pin. Will only know about sdcard when board is powered up with sd card in it, or on M21

I just tested in the new version, nothing changes. here is the settings

[config 2.1.zip](https://github.com/MarlinFirmware/Marlin/files/9803369/config.2.1.zip)

calcdas, 2022-10-21:
Cards with tft that do not have the encoder do not open the sdcard browser. The problem is in the menu_media

Andrzej220754, 2022-10-21:
I will try. Thank you

calcdas, 2022-10-23:
I find it funny how they like to close the discussions, without solving the problems there are several with the same problem since they released the version with tft touch.
just one more note in extui dir has anycubic_chiron , see how it uses filenavigator to access sdcard files without needing encoder

lipewww, 2022-11-01:
why don't you send some solution options? I've been unable to use an sd card for over 6 months.

Andrzej220754, 2022-11-03:
I changed to 2.09

closed 2023-01-13 by github-actions[bot]