MarlinFirmware/Marlin #24892: [BUG] (bug summary)

Dtcreation, 2022-10-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Impossible to flash Tronxy X5SA Pro v6 Board

### Bug Timeline

first Atempt

### Expected behavior

Flashing the board with the SD card with the update.cdb file on it

### Actual behavior

Nothing happens

### Version of Marlin Firmware

2.1.1

### Printer model

Tronxy X5SA Pro

### Electronics

Board v6

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/9801617/Configuration_adv.zip)
[X5SAPro-446+Titan.zip](https://github.com/MarlinFirmware/Marlin/files/9801619/X5SAPro-446%2BTitan.zip)
Here firmware files from Tronxy (last official firmware)

ellensp, 2022-10-18:
Please update the subject line to something more useful than "[BUG] (bug summary)"

xC0000005, 2022-10-18:
If the printer is one of the ones that uses the "update.cbd" format, set your baud to 115200, 8, N, 1, put hte card in and reboot. Paste the output from the bootloader into here.

Baaziak, 2022-10-24:
I have another printer (Tronxy XY-3 SE), but the same board and the same problem.
Yes, it's updated with update.cbd
That's what i get if i use chitu_f103 build task:
"start
External Reset
Marlin bugfix-2.1.x
echo: Last Updated: 2022-10-01 | Author: (none, default config)
echo: Compiled: Oct 23 2022
echo: Free Memory: 11143  PlannerBufferBytes: 2048
echo:EEPROM version mismatch (EEPROM=? Marlin=V87)
echo:Hardcoded Default Settings Loaded
echo:SD card ok"

And it keeps repeating.
At least in version 2.0.9 this problem already existed. The problem only applies to the LVGL interface, everything is OK with Classic and Color.

If You use chitu_f103_maple build task, no problems with the build, even the LVGL interface. But to use _maple build task you need to make some manipulations with the stm32f1-maple.ini file. If necessary, I can check what I changed there. 
And here is the answer when building with _maple:
"start
PowerUp
Marlin bugfix-2.1.x
echo: Last Updated: 2022-10-01 | Author: (none, default config)
echo: Compiled: Oct 24 2022
echo: Free Memory: 4063  PlannerBufferBytes: 2048
echo:V87 stored settings retrieved (791 bytes; crc 44724)
echo:SD card ok"

ellensp, 2022-10-24:
There are at least two V6 boards
The original TronXY CXY-V6 board with a STM32F103ZE
And the newer TronXY CXYV6-191121 with a STM32F446ZE 

The latter has not been added into marlin yet.

ellensp, 2022-10-24:
These newer boards reportedly do not use  update.cdb they use a non encrypted bin file.
from: https://tronxy.fandom.com/wiki/TronXY_Mainboard_Documentation#TronXY_CXYV6-191121/CXYV10-20210309_(STM32F446_based)

Baaziak, 2022-10-24:
I have V6, based on STM32F103. And I uploaded some chitubootloader.bin to my chip so that it can install the firmware from update.cbd.
Is it normal that the amount of free memory on a chip depends so much on the build task configuration (chitu_f103 / citu_f103_maple)?

The-EG, 2022-10-24:
> Is it normal that the amount of free memory on a chip depends so much on the build task configuration (chitu_f103 / citu_f103_maple)?

Yes, they are completely different frameworks. maple takes up about 20K more flash on my setup if I recall correctly. Maple should only be used in cases when the other environment doesn't work for some reason.

ellensp, 2022-10-24:
@Baaziak your issue is not related to the issue the original poster made

```
Expected behavior
Flashing the board with the SD card with the update.cdb file on it

Actual behavior
Nothing happens
```

Ie it does not flash at all,  due to it being the f446

It is not the same board or the same symptoms 

Please don't co-opt this issue for your unrelated issues..

Dtcreation, 2022-10-24:
My board is the f446 (CXYV6-191121), will i be added to Marlin ?

Thanks

closed 2023-01-03 by github-actions[bot]