MarlinFirmware/Marlin #24894: [FR] (feature summary)

tizi24, 2022-10-17:
### Are you looking for hardware support?

add support for 0327001 v2.0 mks custom board and At32f403argt7 ARM microcontroller please!!!

### Describe the feature you want

add support for 0327001 v2.0 mks custom board and At32f403argt7 ARM microcontroller please!!!

ellensp, 2022-10-18:
Please update the subject line to something more useful than "{FR] (feature summary) "

How about some details?
What printer was this found in?
What about some high res in focus pictures of the motherboard in question?

tizi24, 2022-10-18:
> Actualice la línea de asunto a algo más útil que "[FR] (resumen de características) "
> 
> ¿Qué tal algunos detalles? ¿En qué impresora se encontró esto? ¿Qué pasa con algunas imágenes enfocadas de alta resolución de la placa base en cuestión?

Hello, I found it in a printer called hellbot Magna se 32 bit, but the board says Wanhao that it is a custom board made by Makerbase, I have a supposed Firmware but you have to compile with IAR TOOLS, according to the people of makerbase they say that the Firmware is correct that they passed me but Configuration.h data is missing the microcontroller is an ARM [AT32F403ARGT7[]
[image]
[image]
[image]
[image]
[image]
[image]

ellensp, 2022-10-18:
Can you attach firmware they provided? It may still hold clues

tizi24, 2022-10-18:
> 
yes, it weighs 150 mb https://wetransfer.com/downloads/3a0655e7ec868528f360f89c56adc41c20221011162218/82fc54?fbclid=IwAR1xjsFXBI5pzSFWarnJ11HCRssTegU7azaZXce7gNYkKvAi7f9Lq2DnMN0

ellensp, 2022-10-18:
all the board stuff is  external to marlin... 

Seem to support 5 boards
Middlewares/Third_Party/Marlin/pins_Lite.h
Middlewares/Third_Party/Marlin/pins_Robin.h
Middlewares/Third_Party/Marlin/pins_Robin2.h
Middlewares/Third_Party/Marlin/pins_RobinNano.h
Middlewares/Third_Party/Marlin/pins_RobinPro.h

tizi24, 2022-10-18:
> all the board stuff is external to marlin...
> 
> Seem to support 5 boards Middlewares/Third_Party/Marlin/pins_Lite.h Middlewares/Third_Party/Marlin/pins_Robin.h Middlewares/Third_Party/Marlin/pins_Robin2.h Middlewares/Third_Party/Marlin/pins_RobinNano.h Middlewares/Third_Party/Marlin/pins_RobinPro.h

yes, it's too confusing, my intention is to be able to make it natively compatible with marlin and eliminated the firmware project that they sent me that I can't understand how to define the board in Configuration.h

ellensp, 2022-10-18:
they don't seem to do that...

in pins.h
```
#if defined(MKS_ROBIN_LITE)		//robin_lite?�¡�?
  #include "pins_Lite.h"
#elif defined(MKS_ROBIN) 		//MKS Robin?�¡�?
  #include "pins_Robin.h"
#elif defined(MKS_ROBIN_MINI)	//MKS RobinMini?�¡�?
  #include "pins_RobinMini.h"
#elif defined(MKS_ROBINPRO)		//MKS RobinPro?�¡�? (?����?F407?�¡�?)
  #include "pins_RobinPro.h"
#elif defined(MKS_ROBIN2)		//MKS Robin2?�¡�?
  #include "pins_Robin2.h"
#elif defined(MKS_ROBIN2MINI)	//MKS Robin2Mini?�¡�?
  #include "pins_Robin2Mini.h"
#elif defined(MKS_ROBIN_NANO)	//MKS RobinNano?�¡�?
  #include "pins_RobinNano.h"
#endif

```

So they are probably just passing a -D at compile time
eg compiler_command -DMKS_ROBIN2
Instead of using a #define MOTHERBOARD

tizi24, 2022-10-18:
> they don't seem to do that...
>
> in pins.h
>
> ```
> #if defined(MKS_ROBIN_LITE)		//robin_lite?�¡�?
>   #include "pins_Lite.h"
> #elif defined(MKS_ROBIN) 		//MKS Robin?�¡�?
>   #include "pins_Robin.h"
> #elif [...]

is there no way to make it native to marlin and get away from patches to make it work with this board and IAR TOOLS?

tizi24, 2022-10-18:
This is the original firmware delivered by the representative of the 3d printer brand, I understand that the opne source firmware delivered, compiles according to the chosen model and then with the file "robin_nano_cfg.txt" certain parameters of the printer are adjusted and of what is shown on the screen

[Firmware-Magna-SE-32-bits-2.4.2.zip](https://github.com/MarlinFirmware/Marlin/files/9807746/Firmware-Magna-SE-32-bits-2.4.2.zip)

ellensp, 2022-10-18:
found this https://github.com/martinloren/Artery-AT32-PlatformIO  
Adds support for this processor to platformio at least in theory.  This is a good start.

But it doesn't add a arduino frame work for this processor..  so someone would need to write/port this part

tizi24, 2022-10-18:
> found this https://github.com/martinloren/Artery-AT32-PlatformIO and https://github.com/ArteryTek/platform-arterytekat32 Adds support for this processor to platformio at least in theory. This is a good start.
> 
> But it doesn't add a arduino frame work for this processor.. so someone would need to write/port this part There is https://github.com/ArteryTek/arduino-core-at32 but sadly it is empty.

well let's hope we're on the right track, if at least I could get what board is defined in the compilation with the board it would already be an advance

Guilleas21, 2023-02-27:
> > 
> 
> yes, it weighs 150 mb https://wetransfer.com/downloads/3a0655e7ec868528f360f89c56adc41c20221011162218/82fc54?fbclid=IwAR1xjsFXBI5pzSFWarnJ11HCRssTegU7azaZXce7gNYkKvAi7f9Lq2DnMN0

Me podrás volver a pasar esto? Es el firmware para está placa ? La mía es de 8 bit ,sirve igual ?

tizi24, 2023-02-27:
> Busco firmware para está placa que adjunto ,la tengo en una magna se de 8 bit ,tengo que editar el pin ,pero no lo consigo [image]

El firmware base se lo podes pedir a la gente de Helltbot, sino hay un grupo de whatsapp/telegram llamadao "Buena junta 3D"  buscalo por youtube

Guilleas21, 2023-02-27:
> bit ,tengo que editar el pin ,pero no lo consigo [image:
> IMG_20230223_154800]

Buenas ,no lo tiene nadie y los de hellbot no te lo dan

tizi24, 2023-02-28:
> […](#)

aca te lo paso https://drive.google.com/file/d/1cu7Pp-phAq2_In1EZ8LCnz8ooiNQfiy_/view?fbclid=IwAR37yJMsTK5aWz2fk9bbxbeuva4bSue-gWqs0ktYT1RJ1Kdocn5hcQIxcJk

tizi24, 2023-02-28:
queda cerrado

closed 2023-02-28 by tizi24

Guilleas21, 2023-02-28:
Uuu muchas gracias!!!!!!! Cómo lo buscaste ? ? Tenés el modelo de la placa ?

Guilleas21, 2023-02-28:
por casualidad tendras el datasheet de la placa?

Guilleas21, 2023-02-28:
Capas me podés ayudar ,se me quemó la salida del fan de capa ,y quería
cambiarla por la que está usando el disipador del hotend ,pero tengo que
editar el pin ,dónde lo puedo encontrar a eso en marlin ? Y si sabes.como
figuraría ... gracias!!!