MarlinFirmware/Marlin #24900: faulty home switch causes machine to damage itself

jbhome, 2022-10-18:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I purchased a small CNC machine using dual endstops on both X and Y axis. In testing it, I issued a home command. Due to incompatibility between the limit switch connector and the control board the software saw one limit switch closed and one open. This caused one motor to home and the other to not run. This in turn caused the mechanism to twist until it broke the carriage mechanism.
I would like to suggest adding a max steps during homing to avoid this. In other words, if one motor stopped and the other has not homed within x steps, stop both motors and report a fault.
I think this would be beneficial for both X and Y.

### Expected behavior

I expected the mechanism to move towards home, with both motors running until the associated limit switches operated.

### Actual behavior

One motor ran continuously, eventually stalling while the other motor did not run at all due to the false at home state seem by the controller.

### Steps to Reproduce

Disconnect on home switch
Initiate axis home from host software such as pronterface.
Observe motor with disconnected limit switch does not move and is in fact locked at current position while the other motor tries to drag the whole mechanism home. It only succeeds in moving it's side toward home until whatever compliance in the mechanism is consumed, and then stalls. In my case as the carriages are printed PLA, this resulted in both carriages splitting in half.

### Version of Marlin Firmware

Not sure. Suspect this is a common "Feature" 

### Printer model

V1 Engineering MPCNC Primo

### Electronics

SKR Pro V1.2

### Bed Leveling

No Bed Leveling

### Your Slicer

Other (explain below)

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I have not included a config file. While there have been config changes by V1 Engineering, I think this issue is internal to the Marlin firmware and not something the user has direct control of via the config file or otherwise.

thisiskeithb, 2022-10-18:
> there have been config changes by V1 Engineering

What else did they modify? We can’t support forks of Marlin here.

> I would like to suggest adding a max steps during homing to avoid this

Make sure you have `VALIDATE_HOMING_ENDSTOPS` enabled since that’s exactly what it does. Marlin will halt if endstops don’t trigger in the steps required to home an axis` max length.

`DETECT_BROKEN_ENDSTOP` is another option I’d recommend you enable.

closed 2022-10-18 by thisiskeithb

ellensp, 2022-10-18:
When testing end switches, you test the end switches manually by hand with the M119 gcode . not with homing. 
Your method is like testing new car breaks by accelerating at a wall and then trying the breaks.

jbhome, 2022-10-18:
Thanks Keith. I'll check with v1 to make sure they are doing this.
Now om to  calibrating my ender to cobratee it so I can print replacement
parts. Argh.
Have a great day
Jim