MarlinFirmware/Marlin #24901: [FR] (disable homing independent axis's X, X2, Y, Y2, Z0, Z1, Z3, Z4)

alfskaar, 2022-10-19:
### Is your feature request related to a problem? Please describe.

I'm working on implementing ERCF, ENRAGED RABBIT CARROT FEEDER on local HW using BTT Octopus Pro this is on Voron 2.4 R2
I'm using:
X
Y
Z0 $ Z1 (hopeful BTT will make a future version where this two are not on parallel stepper driver)
Z2
Z3
A (I) selector ERCF
B (J) gear that load and unload filament
Servo (pins_BTT_OCTOPUS_V1_common.h SERVO0_PIN PB6)

It will be useful to treat any axis's that is not X, X2, Y, Y2, Z0, Z1, Z3, Z4 as independent axis's

disable homing I
disable homing J

I'm use G28, G34 this also home A, B this is not needed as A is the selector and will only move on filament change, B is the gear (extruder) that load filament and this is disengage with the servo and filament is just running free going true it.

#define MAIN_MENU_ITEM_1_DESC "Home XYZ & Z stepper auto align"
#define MAIN_MENU_ITEM_1_GCODE "G28 XYZ\nG34 W"

I use StallGuard to home A this is already working. ERCF have end stop but I'm not using it since it will be useful to have StallGuard if any filament is in the way or some how is locking up the selector.

Next to work on is to home B (load filament using StallGuard)
Under the gear at the extruder I have FILAMENT_RUNOUT_SENSOR "#define FIL_RUNOUT_PIN PG12" so is loading fails I think this should detect it and try again.

github.com/EtteGit/EnragedRabbitProject

### Are you looking for hardware support?

.

### Describe the feature you want

.

### Additional context

.

ellensp, 2022-10-19:
I think your potentially doing this the wrong way.  Ie your basically trying to implement this without any code...

You have told marlin you have a I and a J so its treating it like a standard axies

You need to actually add code ...  

Something like defines to say which extruders are what

#define ERCF_selector_on E3
#define ERCF_loader_on  E4

with new code that moves these steppers as needed on tool change

Have your thought about the slicing?

with your method your going to need to add gcodes to manually move I and J
vs a simple Tool change command.

alfskaar, 2022-10-19:
I have be thinking about slicing yes but the problem is that StallGuard is not available if I defined them as extruder so as a prof of concept I doing it to see if the HW can do this then I will add the need code. I have to start somewhere. So I want to be sure my idea can work before going to different FW.

ellensp, 2022-10-19:
You should take a look at https://github.com/MarlinFirmware/Marlin/issues/24196 for hot to disable the end stops in code.

DerAndere1, 2023-08-06:
Problem 1) With the recent endstop refactoring, I do not know how to disable endstops for an axis. Maybe undefine `I_MIN_PIN` ?

Problem 2) There is no solution for an infinite axis when HAS_POSITION_SHIFT is defined (CNC_COORDINATE_SYSTEMS enabled and/or NO_WORKSPACE_OFFSETS disabled): With HAS_POSITION_SHIFT,  G92 modifies variable `position_shift` and thus does not prevent overflow of the variable `current_position`. For  a possible fix for problem 2, see  https://github.com/Snapmaker/Snapmaker2-Controller/commit/65c7e67ad42081a9656985bc8c7630543d39cec2

thisiskeithb, 2024-03-02:
https://github.com/MarlinFirmware/Marlin/pull/26174 has been merged.

closed 2024-03-02 by thisiskeithb