MarlinFirmware/Marlin #24904: [BUG] SWITCHING_EXTRUDER Servo movement fails

jesterhead82, 2022-10-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Toolchange (either by Tx, or by using LCD Menu entry in MOTION section), does not trigger servo movement.

Setup

- Single Nozzle, 2 Extruder setting using shared E0
- Servo to Switch Extruder gears

Manual movement of Servo (e.g. M280 P1 S5) works as expected, Extruder direction changes also with Toolchange.

As a workaround I added manual Servo movement to EVENT_GCODE_TOOLCHANGE_T0 / EVENT_GCODE_TOOLCHANGE_T1

### Expected behavior

Servo Moves to T0 after start of printer and switches during T0/T1 change

### Actual behavior

No movement triggered by Toolchange, no init of Position (servo is powerless) after startup

### Version of Marlin Firmware

Bugfix 2.1.x as of this week

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/9823757/Config.zip)

ellensp, 2022-10-20:
Provided config does not compile .. to many recent changes to bugfix config files
Servo does move on T0 and T1 as expected 
Servo does not initialize to T0 position on startup

ellensp, 2022-10-20:
Provided config does not compile .. to many recent changes to bugfix config files

Updated configs [updated.zip](https://github.com/MarlinFirmware/Marlin/files/9826532/updated.zip)

Servo does move on T0 and T1 as expected 
Servo does not initialize to T0 position on startup

ellensp, 2022-10-20:
weird this movement to position 8 ish is not coming from anywhere obvious... 
At the moment I have it so that T0 or t1 moves correctly, but then something I cannot locate moves back to the weird location

ellensp, 2022-10-20:
gah you have 
  #define EVENT_GCODE_TOOLCHANGE_T0 "M280 P1 S25" // Extra G-code to run while executing tool-change command T0
  #define EVENT_GCODE_TOOLCHANGE_T1 "M280 P1 S6"      // Extra G-code to run while executing tool-change command T1

jesterhead82, 2022-10-20:
Hi,

I'm working on a switching extruder where I use a Toothed Rack to press L/R idler against the driveshaft, therefor I only need some degrees of movement on the servo.

As for Event_GCode, i'm misusing that that as a workaround, behaviour was the same for me without it.

ellensp, 2022-10-20:
in Marlin/src/module/tool_change.cpp
Find
`    if ((EXTRUDERS & 1) && e < EXTRUDERS - 1) {`
replace it with:
`    if (e < EXTRUDERS) {`

This works as expected, but im still working out why it was as it was....

jesterhead82, 2022-10-20:
Will test, but will take 2 weeks.

Btw could be because of a 2x2 Setup (nozzle and extruder switch, there's an BR for that setup where some stuff has been modified)

ellensp, 2022-10-20:
A PR has been created to fix this.

closed 2022-10-20 by ellensp

jesterhead82, 2022-11-01:
I want only to confirm that the srvo is working now, thanks.