MarlinFirmware/Marlin #24907: [BUG] Compiler Warnings in VScode for SKR E3 mini v1.2, bugfix-2.1.x

fnsign, 2022-10-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Since the last few bugfix-2.1.x releases I receive following two compiler warnings:
config files are attached as zip.

Warning 1:
```
[{
	"resource": "/Users/fozi/Documents/MarlinFirmware/NightlyBuilds/Marlin-1/Marlin/src/gcode/feature/advance/M900.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "unused variable 'tool_index' [-Wunused-variable]",
	"startLineNumber": 52,
	"startColumn": 23,
	"endLineNumber": 52,
	"endColumn": 23
}]
```

Warning 2:
```
[{
	"resource": "/Users/fozi/Documents/MarlinFirmware/NightlyBuilds/Marlin-1/Marlin/src/libs/duration_t.h",
	"owner": "cpp",
	"severity": 4,
	"message": "'%02hu' directive writing 2 bytes into a region of size between 0 and 4 [-Wformat-overflow=]",
	"startLineNumber": 161,
	"startColumn": 30,
	"endLineNumber": 161,
	"endColumn": 30
}]
```

### Bug Timeline

Warnings appeard about 2 weeks ago

### Expected behavior

Compilation runs through without warnings.

### Actual behavior

Above mentioned compiler warnings appear in VScode

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality Ender 3 Pro

### Electronics

SKR Mini E3 v1.2

### Bed Leveling

ABL Bilinear mesh

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/9827174/configs.zip)

GR8YT1, 2022-10-22:
Same two errors, 2.1.x, Ender 3, SKR Mini E3 V3.

Got rid of the first one by changing line 52 in M900.cpp
original line: constexpr uint8_t tool_index = 0;
Revised line:  __attribute__ ((unused)) constexpr uint8_t tool_index = 0;

Found that solution searching, no idea if it is the correct way to handle it, but it did get rid of the warning.  Still playing with the -Wformat-overflow warning

EvilGremlin, 2022-10-22:
overflow is of no consequence (and i fixed it already, will be merged when i tidy up some other stuff)

closed 2023-01-01 by github-actions[bot]