MarlinFirmware/Marlin #24916: [BUG] ProUI: Parts of bootsplash rendered behind text

Matombo, 2022-10-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Parts of the bootsplash are rendered behind text, most notable in the main menu:
[image]

### Expected behavior

The space behind text is a solid color.

### Actual behavior

Parts of the bootsplash are rendered behind text.

### Version of Marlin Firmware

2.1.1

### Printer model

Ender 3 V2

### Electronics

Mainboard 4.2.2 with GD32F303, DACAI LCD (the one using the "private" firmware folder)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9844253/Configuration.zip)

Matombo, 2022-10-22:
Couldn't test bugfix branch as it will not boot.

Forget to add: Because of the GD chip, I have to use the maple target, otherwise Marlin will not boot as soon as I enable BL Touch.

ellensp, 2022-10-22:
There are 3 known different types of this display
The original DWIN the first clone DACAI the second clone SWM32SRET5
Need to identify what one you have...
https://marlin.crc.id.au/forum/t/identifying-ender-3-v2-screen-versions/960

Matombo, 2022-10-22:
I did in the Electronics section:
DACAI LCD (the one using the "private" firmware folder)

Matombo, 2022-10-22:
I put it in a more prominent position.

mriscoc, 2022-11-18:
That could be a font problem, try with https://github.com/mriscoc/Ender3V2S1/wiki/How-to-update-the-display for Marlin change the icon set library.

Matombo, 2022-12-03:
Self compiled marlin version only work for me when using this icon set: https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-3%20V2/LCD%20Files
This icon set however breaks the ui for any precompiled marlin FW

thisiskeithb, 2022-12-03:
> Self compiled marlin version only work for me when using this icon set

> This icon set however breaks the ui for any precompiled marlin FW

That is to be expected because you need to use the icon set that matches your firmware. They’re not backward / cross-compatible to other firmware versions.

> Mainboard 4.2.2 with GD32F303

> Couldn't test bugfix branch as it will not boot.

We do not support GD32 at this time, but we have https://github.com/MarlinFirmware/Marlin/discussions/24985 to track process.

closed 2022-12-03 by thisiskeithb