MarlinFirmware/Marlin #24922: [BUG] ADAPTIVE_STEP_SMOOTHING on SKR mini E3 V3 - IRQ Priorities

rileymcdowell, 2022-10-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a Big Tree Tech SKR mini E3 V3 mainboard that I am using with Ender 3 Max printer hardware including a BL-Touch probe. This hardware all worked fine with the stock v4.2.2 creality board, but I get intermittent failures with the BL-Touch v3.1 probe on the new mainboard.

The issue I have is described in a [now closed issue](https://github.com/MarlinFirmware/Marlin/issues/23958). The workaround was to disable ADAPTIVE_STEP_SMOOTHING. That issue is closed and locked, so I created a new one to re-open the discussion. 

A major benefit of the BTT SKR mini E3 V3 board is that it has quiet stepper drivers. Stepper volume is further reduced with ADAPTIVE_STEP_SMOOTHING turned on, motivating someone to find a way to enable the feature alongside a BL-Touch probe. In fact, ADAPTIVE_STEP_SMOOTHING works fine, but the BL-Touch probe does not work reliably when ADAPTIVE_STEP_SMOOTHING is enabled.

It looks like this same problem happened about 2 years ago with [older versions of the SKR mini E3 board](https://github.com/MarlinFirmware/Marlin/issues/18598). The root cause on the older board was that interrupts from the BL-Touch probe were of lower priority than step and temp timers, which together issue frequent interrupts when ADAPTIVE_STEP_SMOOTHING is enabled. This caused the BL-Touch probe's stop signals to be delayed. The solution was to modify the interrupt priorities [as done here](https://github.com/MarlinFirmware/Marlin/pull/18637) for the STM32F1 board to make sure the probe's interrupts were processed at the highest priority.

Because the SKR mini E3 V3 uses a different MCU (STM32G0) than its predecessor (STM32F1), the fix from that previous pull request doesn't work on the new version of the board.

I tried playing with the IRQ priorities and found a workaround that allows these features to both be enabled. Within the current stm32g0.ini file, the build flags define `-DSTEP_TIMER_IRQ_PRIO=0`. This sets the step timer to maximum priority, above the servo priority that interacts with the BL-Touch probe and causing delayed readings. By setting `-DSTEP_TIMER_IRQ_PRIO=1`, the servo and step timers are given equal priority, and I get acceptable performance from my probe while ADAPTIVE_STEP_SMOOTHING is enabled.

I created this issue to share this with the Marlin community. The consensus seems to be that the STM32G0 MCU is not capable of running ADAPTIVE_STEP_SMOOTHING and a BL-Touch probe simultaneously. In fact, a [change was merged](https://github.com/MarlinFirmware/Configurations/pull/727) to disable the configuration on the Ender 3 V2. 

There is likely a better way to solve this problem than modifying build flags, but it seems to me it may just be an IRQ priority issue rather than a lack of capability in the MCU.

### Bug Timeline

Begins with the release of SKR mini E3 V3

### Expected behavior

I expect Marlin's configuration for the STM32G0 MCU to set an appropriate IRQ priority among step, servo, and temp interrupts so that the BL-Touch probe works reliably when ADAPTIVE_STEP_SMOOTHING is enabled.

### Actual behavior

I get inconsistent BL-Touch probe readings, sometimes delayed as much as a milimiter or more. If an auto-home probe is too delayed, the nozzle can impact the bed.

### Steps to Reproduce

1. Comple the bugfix-2.1.x branch with ADAPTIVE_STEP_SMOOTHING and BLTOUCH enabled targeting the STM32G0B1RE_btt platform.
2. Flash a printer with the BTT SKR E3 mini V3 mainboard and a BL-Touch V3.1 probe
3. Attempt to build a UBL mesh (or anything else that does a bunch of probes)
4. Observe the probe - you can see it's inconsistent
5. Alternatively, visualize the resulting mesh - it will have many points that are outliers due to failed probes.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality Ender 3 Max

### Electronics

BigTreeTech SKR mini E3 V3

### Add-ons

BL-Touch

### Bed Leveling

UBL Bilinear mesh

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

lovebug1993, 2022-10-23:
Hi my name is McDowell

rileymcdowell, 2022-10-23:
[stm32g0.zip](https://github.com/MarlinFirmware/Marlin/files/9847609/stm32g0.zip)

Here's a copy of the stm32g0.ini file that I used to compile firmware that worked for me.

lovebug1993, 2022-10-23:
Why do you keep emailing me?

rileymcdowell, 2022-10-23:
> Why do you keep emailing me?

I'm not emailing you, I just opened an issue on the Marlin Firmware Github repository. I am guessing you are subscribed to some set of notifications for that repository and it's sending you email notifications. If that's not something you want, I'm sure you can poke around and find a way to unsubscribe.

thisiskeithb, 2022-10-24:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/24351

closed 2022-10-24 by thisiskeithb

rileymcdowell, 2022-10-24:
@thisiskeithb 

Just to confirm, you (commented)[https://github.com/MarlinFirmware/Marlin/issues/24351#issuecomment-1154562800] in #24351 

> Disabling ADAPTIVE_STEP_SMOOTHING seemed to remedy this issue when this bug was reported here before (see #23958), but I see it is disabled in your config.

It sounds like I am a duplicate of #23958, but not #24351.

#23958 concluded that this configuration cannot work, but I have it working using the `-DSTEP_TIMER_IRQ_PRIO=1` build flag. It would be nice if the end of the comments chain on that closed issue somehow linked to this information for anyone who wants to get it working.

thisiskeithb, 2022-10-24:
Please comment in https://github.com/MarlinFirmware/Marlin/issues/24351