MarlinFirmware/Marlin #24925: [FR] Long file names for Octoprint and similar controllers

etherbrat, 2022-10-24:
### Describe the feature you want

Octoprint cannot list file names in other than 8.3 format if Marlin does not provide them. It appears that the problem is ultimately attributable to the faux serial (via USB) interface between the printer and Octoprint, which (strictly) has no capacity for anything other than 7-bit ASCII. The Octoprint team has been staunchly resisting work-arounds for years! This is a fundamental problem for the current form of the USB interface, but it is frustrating to see long file names on my printer's LCD but not in Octoprint. Is there a way that Marlin could be configured to output long file names in (say) BASE64, allowing for both long file names and Unicode? e.g. an "M" code to switch to BASE64, which would just fail (or negatively acknowledge) if the printer does not have BASE64 support? On a more futuristic note, I hope and assume that Marlin has a better way of sending file lists to Octoprint when the printer has built-in WiFi and all the other trimmings, e.g. TCP/IP, DHCP, AVAHI, SMB, etc. My printer is a humble Wanhao i3 Mini, so I wouldn't know.

thisiskeithb, 2022-10-24:
This already exists as `LONG_FILENAME_HOST_SUPPORT` and can be retrieved with `M20 L`

closed 2022-10-24 by thisiskeithb

mehmetsutas, 2022-11-06:
@thisiskeithb I think something have changed I have upgraded to latest bugfix from 2.0.9.3. In my previous setup M20 L was reporting the full filename along with the path. With the latest version only the first 13 characters of the filename is reported even when the LONG_FILENAME_HOST_SUPPORT option is enabled.