MarlinFirmware/Marlin #24928: [FR] Add a switch to M593 to disable input shaping

tombrazier, 2022-10-25:
### Is your feature request related to a problem? Please describe.

@InsanityAutomation has suggested an extension to M593 to disable input shaping. Whilst input shaping is effectively has no effect when damping ratio is equal to 1, all the logic is still being called and the generated pulse train is ever so slightly different. I think being able to disable the feature completely is worthwhile.

### Describe the feature you want

Here is what I think is needed for this feature:

- A boolean in `class Stepper`.
- A new flag for M593.
- An if statement around https://github.com/MarlinFirmware/Marlin/blob/6b4d7b9151d260d3109ca857d084e3ca0cd8b048/Marlin/src/module/planner.cpp#L2487-L2491
- An if statement around https://github.com/MarlinFirmware/Marlin/blob/6b4d7b9151d260d3109ca857d084e3ca0cd8b048/Marlin/src/module/stepper.cpp#L1639-L1643
- An if statement around https://github.com/MarlinFirmware/Marlin/blob/6b4d7b9151d260d3109ca857d084e3ca0cd8b048/Marlin/src/module/stepper.cpp#L1822
- An if statement around https://github.com/MarlinFirmware/Marlin/blob/6b4d7b9151d260d3109ca857d084e3ca0cd8b048/Marlin/src/module/stepper.cpp#L2497-L2530

Additionally a few CPU cycles could be saved with if statements around other code that is conditional on `INPUT_SHAPING` or `HAS_SHAPING_X` or `HAS_SHAPING_Y`.

The-EG, 2022-10-25:
Instead of adding a new parameter to `M593`, disabling an axis could be accomplished by specifying a frequency of 0. This is the existing behavior for RRF.
It may help to be consistent, I think, or would that be confusing for some reason?

InsanityAutomation, 2022-10-25:
I'd agree is 0 should be effectively off, instead of comparing the bool we could compare for value over 0.

tombrazier, 2022-10-25:
That would save the space of a bool. It would also mean axes can be switched off independently. The code changes would be a bit different to what I said above. Actually, that would be simple enough that I can throw a PR together really quickly.

tombrazier, 2022-10-25:
New branch https://github.com/tombrazier/Marlin/tree/is_improvements has freq = 0 to switch off input shaping. I'll add bug fixes for TMC2208 here too hopefully.

gudvinr, 2022-12-20:
This probably should be closed since PR was merged

closed 2022-12-20 by ellensp