MarlinFirmware/Marlin #24929: [BUG] Double z mode with asymmetric Z-axis step

amxsa, 2022-10-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

[BUG] Double z mode with asymmetric Z-axis step

MKS GEN_L V2.1 ，Z2_SERIAL_TX_PIN  and Z2_SERIAL_RX_PIN not define ？
```
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define Z2_DRIVER_TYPE TMC2209
#define E0_DRIVER_TYPE TMC2209
```

### Expected behavior

Double z mode, Z axis step symmetry

### Actual behavior

Double z mode with asymmetric Z-axis step

### Version of Marlin Firmware

marlin 2.1.1

### Printer model

MKS GEN_L V2.1

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9871987/Configuration.zip)

amxsa, 2022-10-26:
[image]

In fact, the Z-axis is in uart mode, but the program prompts a TMC connection error

ellensp, 2022-10-27:
The uarts pins should auto allocated from the unused E, this is a bug...
The TMC error is probably a bad stepper driver.

closed 2022-10-27 by ellensp