MarlinFirmware/Marlin #24937: [FR] Add reverse sort of file names on SD

vovodroid, 2022-10-28:
### Is your feature request related to a problem? Please describe.

Currently files could be sorted by name, with option recent FAT entry first. So it could difficult fine new file.

### Describe the feature you want

Some (at least Prusalicer) allows add date/time at the file name beginning like 202221028-174608. Reverse sort puts new file at the list top.

### Additional context

Actually I build Marlin with such sort. Just changed 

`#define _SORT_CMP_NODIR() (strcasecmp(sortnames[o1], sortnames[o2]) > 0)`
to
`#define _SORT_CMP_NODIR() (strcasecmp(sortnames[o1], sortnames[o2]) < 0)`
in Marlin/src/sd/cardreader.cpp

I can add config option and create PR if it seems reasonable.

GMagician, 2022-11-30:
Don't expect an answer. I suggest you to create PR anyway and wait

vovodroid, 2022-11-30:
PR #25038

closed 2022-11-30 by thisiskeithb