MarlinFirmware/Marlin #24945: [FR] (feature summary)

classicrocker883, 2022-10-31:
### Describe the feature you want

### Input Shaping Menu
for editing the values through the LCD.

in file --> **lcd/e3v3/proui/dwin.cpp**

under `void Draw_Prepare_Menu()`  add

>#if ENABLED(SHAPING_MENU)
      MENU_ITEM(ICON_Language, MSG_INPUT_SHAPING, onDrawSubMenu, Draw_InputShaping_Menu);
    #endif

this following was below `#endif // HAS_TOOLBAR` added above `// Special Menuitem Drawing functions ===========`

```
// M593 - Acceleration items (Input Shaping)

#if ENABLED(SHAPING_MENU)

void ApplyFreq() { stepper.set_shaping_frequency(HMI_value.axis, *MenuData.P_Float ); }
void ApplyZeta() { stepper.set_shaping_damping_ratio(HMI_value.axis, *MenuData.P_Float ); }

void SetXFreq() { HMI_value.axis = X_AXIS, SetPFloatOnClick(1.0f, 200.0f, 2, ApplyFreq); }

void SetYFreq() { HMI_value.axis = Y_AXIS, SetPFloatOnClick(1.0f, 200.0f, 2, ApplyFreq); }

void SetXZeta() { HMI_value.axis = X_AXIS, SetPFloatOnClick(0.0f, 1.0f, 2, ApplyZeta); }

void SetYZeta() { HMI_value.axis = Y_AXIS, SetPFloatOnClick(0.0f, 1.0f, 2, ApplyZeta); }

  void Draw_InputShaping_Menu() {
    checkkey = Menu;
    if (SET_MENU(InputShapingMenu, MSG_INPUT_SHAPING, 8)) {
      BACK_ITEM(Draw_Prepare_Menu);
    // M593 F Frequency
    #if HAS_SHAPING_X
      static float xfreq = stepper.get_shaping_frequency(X_AXIS);
      EDIT_ITEM(ICON_MoveX, MSG_SHAPING_X_FREQ, onDrawPFloat2Menu, SetXFreq, &xfreq);
 
    #endif
    #if HAS_SHAPING_Y
      static float yfreq = stepper.get_shaping_frequency(Y_AXIS);
      EDIT_ITEM(ICON_MoveY, MSG_SHAPING_Y_FREQ, onDrawPFloat2Menu, SetYFreq, &yfreq);
    #endif
    // M593 D Damping ratio
    #if HAS_SHAPING_X
      static float xzeta = stepper.get_shaping_damping_ratio(X_AXIS);
      EDIT_ITEM(ICON_MoveX, MSG_SHAPING_X_ZETA, onDrawPFloat2Menu, SetXZeta, &xzeta);
 
    #endif
    #if HAS_SHAPING_Y
      static float yzeta = stepper.get_shaping_damping_ratio(Y_AXIS);
      EDIT_ITEM(ICON_MoveY, MSG_SHAPING_Y_ZETA, onDrawPFloat2Menu, SetYZeta, &yzeta);
    #endif
    }
  UpdateMenu(InputShapingMenu);
  }
#endif
```

### Additional context

this is how I was able to get the Input Shaping Menu to work in the ProUI.

if any changes can be made, let me know. but i tested and it does work.

thisiskeithb, 2022-10-31:
Please submit a PR with these changes so it can be reviewed.

Sonicboom247, 2022-12-27:
How would this work for something like the Artillery Sidewinder? I can never define a TFT and just tried to add a Custom Menu item and it says no TFT is defined, sort of a catch 22

thisiskeithb, 2022-12-27:
> How would this work for something like the Artillery Sidewinder?

This Sidewinder’s TFT is a standalone computer / G-code sender and runs separate firmware.

closed 2023-12-03 by classicrocker883