MarlinFirmware/Marlin #24948: USB-SD_HOST_DRIVE is related with Serial port definition order

DaLiV, 2022-10-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Board : BTT-Octopus Pro
if Serial defined ports in order 1,-1,3  /3-rd port assigned to esp866 header/
 - then USB Drive is accessible remotely on PC
if Serial defined in order 1,3,-1 /virtual com assignment as last point/
 - then USB Drive not passed thru and occurs freezes also as USB-Serial communication dropping/unstable.

against commit b20db2e1
related define lines below.

in Configuration.h ( for serial as 1,-1,3): - Working
#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1
#define SERIAL_PORT_3 3
in Configuration.h ( for serial as 1,3,-1): - Not Working
#define SERIAL_PORT 1
#define SERIAL_PORT_2 3
#define SERIAL_PORT_3 -1

in Configuration.h: related to SD/USB-OTG
#define SDSUPPORT
#define MOTHERBOARD BOARD_BTT_OCTOPUS_PRO_V1_0
#define BAUDRATE 115200
#define BAUDRATE_2 115200
#define BAUDRATE_3 115200

in Configuration_adv.h
#define SD_IGNORE_AT_STARTUP
#define NO_SD_AUTOSTART
//#define SD_MENU_CONFIRM_START
#define USB_FLASH_DRIVE_SUPPORT
#define USE_OTG_USB_HOST
#define SDCARD_CONNECTION ONBOARD
#define MULTI_VOLUME
#define DEFAULT_VOLUME SV_USB_FLASH_DRIVE

### Expected behavior

logical division must not impact to function or must be "documented / noted in config"

### Actual behavior

Communication loss, no access to card

### Version of Marlin Firmware

b20db2e1a30c6fc8a929f25739a2e6d6e2b599e2

### Printer model

Tronxy X5

### Electronics

BTT Octopus Pro

### Your Slicer

Prusa Slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

DaLiV, 2022-10-31:
zip with full configs somehow is not attacheable

DaLiV, 2022-10-31:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/9903586/configs.zip)

DaLiV, 2022-10-31:
Sadly even this also not stable ... 
after some restarts same issue in both configurations
no idea why that worked and what changed except "power-on" count
possible will be some initialisations issues that in bootloader.

thinkyhead, 2022-11-29:
I'm not able to find any obvious connection between serial order and SD host over USB. Of course, the board needs to be able to support mounting of the onboard SD card, because Marlin doesn't really handle that. It's handled at a lower level by code in the bootloader and maybe hardware handshaking on the board.

Maybe you can find a connection by using `M43` and making sure there are no glitches in how we are allocating pins, ports, and timers.

I have a couple of Octopus-something boards here, so if you bug me on Discord, we could do some messing around and maybe solve this thing in tandem with debugging over ST-Link and other direct methods.

closed 2023-02-08 by github-actions[bot]