MarlinFirmware/Marlin #24950: [BUG] Immediate MINTEMP error on T1 with latest bugfix

justin-vt, 2022-11-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

This is just a heads up - I'm getting a fatal MINTEMP error immediately after running M104 T1 on my SKR3 with bugfix-2.1.x. T0 works fine. I recompiled several times with same result.

Rolling back to 70b3166715903aebc015362c7a96ac52cee4f2ed fixed the problem.

### Bug Timeline

bfa5e9b584205a475fcf4b979df03e0a5a8d7749, but it may have been a prior commit last week

### Expected behavior

T1 would should have heated normally

### Actual behavior

Fatal MINTEMP error

### Steps to Reproduce

Execute M104 T1

### Version of Marlin Firmware

bugfix-2.1.x bfa5e9b584205a475fcf4b979df03e0a5a8d7749

### Printer model

Custom Printrbot

### Electronics

BTT SKR 3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/9913782/Archive.zip)

studiodyne, 2022-11-01:
@thinkyhead @Bob-the-Kuhn  @GMagician 

is_preheating is the var  that lock mintemp while MILLISECONDS_PREHEAT_TIME not elapsed
But since 1.1 , does'nt exist anywhere and preheating ' sensible' sensors is DOWN

Copy  'is_preheating' and look in 1.1 , the var is included in the max errors calculation , but not in 2.xxx releases

Preheating is down , errors occurs by tons , mintemp triggered !

studiodyne, 2022-11-07:
I just found the bug , and i've seen that the new bugfix have made the fix ,  the bug was the temp_dir count in temp.h
3 hours wasted , but i'm happy anyway :-)

closed 2022-11-07 by ellensp