MarlinFirmware/Marlin #24954: [BUG] (X Axis Twist Compensation Wizard Prompting to reset Z Probe offset to 0.00 then exiting)

IzzyMan12078, 2022-11-02:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

The X Axis Twist Compensation Wizard does not run all the time.  After a reset or power cycle of the printer, the X Axis Twist compensation Wizard in Config - Advanced - runs, probes four locations on the bed and resets the Z Probe offset to a value and records / retains the values showed in a M423.  However, many times after starting it prompts me to reset my z probe offset to 0.00.  Regardless of which option I choose, it exits to the main menu and either has or hasn't reset the Z Probe Offset to 0.00.  After doing this (either option) - all values for four points showing with M423 are 0.00.  I have to power cycle the printer, or EEPROM init and reset.  It will work once, then not again without a reset.  I have tried numerous scenarios of wiping my ABL mesh, reseting Z Probe offset, etc - to no avail.

### Bug Timeline

As soon as I compiled and installed 2.1.1

### Expected behavior

I expect it to run each time I attempt to run it.

### Actual behavior

It doesn't run correctly

### Steps to Reproduce

1- Select Z Probe Wizard in Config Advanced menu
2- Either it will or wont work correctly??

### Version of Marlin Firmware

2.1.1

### Printer model

Ender 3 SI Pro with upgraded CR-10 Display

### Electronics

Stock except display and TH3D ABL Probe

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/9924559/Configs.zip)

IzzyMan12078, 2022-11-23:
This is still happening - I would like to try th bug fix version but this took a bunch of work to get going, and not sure how to transfer my configs (ramps/pins env etc.) to the bug fix version - Is there a quick way to move those configs without a line by line, file by file comparison?

thinkyhead, 2022-11-24:
> Is there a quick way to move those configs without a line by line, file by file comparison?

You can drop your old configs in place in the new Marlin, change the version in the heading of each config file, and then follow the sanity check errors to update any changed options. Maybe it will require few or no changes…?

thinkyhead, 2022-11-24:
The most recent changes are by @Giuseppe499 and @tombrazier so they may have more experience with this area of the code at the moment.

tombrazier, 2022-11-24:
Under the covers this wizard uses the leveling code. I think the homing / leveling debugging procedure [e.g. here](https://github.com/MarlinFirmware/Marlin/issues/25010#issuecomment-1325485803) would generate useful info.

IzzyMan12078, 2022-11-24:
> You can drop your old configs in place in the new Marlin, change the version in the heading of each config file, and then follow the sanity check errors to update any changed options. Maybe it will [...]

THanks - I'll consider / try it.  I am concerned then that new features etc. wont be in the config.  I think I'll just have to commit myself to trying to start from scratch.  I will try this first and see how it goes

IzzyMan12078, 2022-11-24:
> Under the covers this wizard uses the leveling code. I think the homing / leveling debugging procedure [e.g. here](https://github.com/MarlinFirmware/Marlin/issues/25010#issuecomment-1325485803) would generate useful info.

Soon -  I'll try to get this done this weekend and post the results.

IzzyMan12078, 2022-11-24:
I will add that when this happens, a power cycle of the printer resolves the issue and The wizard then works correctly.
OR
I can send M423 R - zeroing out the values, and the wizard will run.
SO
I am not dead in the water, and still functional, I just assume that this is not how it is supposed to work, and wanted Devs to know what I'm experiencing.  I appreciate what you all do here.
I do realize that these values once set, should not change, unless working on the X-Axis gantry etc.  Ive been fighting the typical Ender issues with ABL so have been running this more than most people ever will.

I will follow up soon with debug details.

IzzyMan12078, 2022-11-28:
This does seem to be resolved in the latest bug fix.  This also resolved an issue I had not mentioned where the print nozzle would drag across the bed between points when using the X Twist Wizard.  The print head now rises in Z before moving from each point.  Thanks

IzzyMan12078, 2022-11-28:
Closing - All set

closed 2022-11-28 by IzzyMan12078