MarlinFirmware/Marlin #24956: Printer goes in Parking mode and stays there - M25 error

lordAnubis, 2022-11-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When f.e I use the CHEP bed levelling gcode; the printer halt at the m25 code with a parking notification ( different then from the past ). According to the previous firmware, I could use the button to continue or use the knop to travel in the menu. 

### Bug Timeline

New 

### Expected behavior

Presses the button and continues

### Actual behavior

Nothing just sits there

### Steps to Reproduce

place m25 in the code

### Version of Marlin Firmware

 Ender3-v4.2.7-Stock-20221103

### Printer model

Creality Ender 3

### Electronics

board 4.2.7

### Add-ons

none

### Your Slicer

Prusa Slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Have no config file

lordAnubis, 2022-11-03:
Why not M79 for parking instead of M25 with using PARK_HEAD_ON_PAUSE enabled.

thisiskeithb, 2022-11-03:
> Have no config file

Surely you have configs to build your current firmware. Please zip them up and attach them.

GMagician, 2022-11-04:
Printing via SD? It may be solved by #24877

GMagician, 2022-11-04:
> Why not M79 for parking instead of M25 with using PARK_HEAD_ON_PAUSE enabled. If I see it correctly there are 3 M code that are being used as parking M0, M25 and M125.
> 
> Is this a regression of #18191?

In some cases M25 (PARK_HEAD_ON_PAUSE enabled) is internally redirected to M125 so M25 and M125 are the same

thisiskeithb, 2022-11-04:
> In some cases, PARK_HEAD_ON_PAUSE enabled, M25 is internally redirected to M125

Who knows. We're just guessing at what OP has enabled since they didn't attach their configs.

> Version of Marlin Firmware
> Ender3-v4.2.7-Stock-20221103

Is this from a fork of Marlin? We cannot support forks of Marlin here, so you'll need to download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

closed 2023-01-27 by thisiskeithb