MarlinFirmware/Marlin #24960: [BUG] Laser features PWM output not working on SKR E3 RRF

XDr4g0nX, 2022-11-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Laser PWM does not seem to work on a BTT SKR E3 RRF.

### Bug Timeline

New issue / 04-11-2022

### Expected behavior

When setting the "Laser Power" from the "Laser Control" LCD menu on anything higher then 0% with "Laser Power:" turned to "On" that the selected SPINDLE_LASER_PWM_PIN pin would provide a 0-5V output signal.

### Actual behavior

The SPINDLE_LASER_PWM_PIN remains at 0V and the laser does not turn on.

### Steps to Reproduce

-Enable "LASER_FEATURE"
-Define "SPINDLE_LASER_ENA_PIN" & "SPINDLE_LASER_PWM_PIN"

### Version of Marlin Firmware

Marlin 2.1.1

### Printer model

Custom Laser Engraver

### Electronics

BTT SKR E3 RRF

### Add-ons

NEJE A40630 Diode Laser

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9937696/Configuration.zip)

thinkyhead, 2022-11-24:
Try enabling `PINS_DEBUGGING` and issue an `M43` command to get a full pins report. Maybe that will show something is in conflict, either a pin or a timer.

aneantisor, 2023-01-05:
Any news about this problem ? i've got similar issue with a bigtreetech octopus board.
I cannot drive the laser properly.
The only way i get it kind of working is with the pulse test. 
it drive it for about 1ms even if i set 999ms. the pwm duty is working because when i change the power, the laser pulse more or less linked to the power i've entered. 
i tried kind of a lot of thing (fan pin, led pin, end-stop pin).
It looks like pwm is set, and just after that, another function reset it to normal GPIO.
Any suggestion ?

cbagwell, 2023-01-05:
Using the ON/OFF menu to test lasers isn't very useful if you're using the default LASER_SAFETY_TIMEOUT_MS of 1 second as it gets immediately turned off.  If you're seeing any activity at all, this is likely what you're running into.

aneantisor, 2023-01-05:
i tried without the safety feature to be sure and then it work in the ON/OFF menu.
But for a print, i had another problem... marlin as updated the way they handle laser Inline gcode.
I just needed to add "I" argument to the M3 start Gcode.
for me, is it solved. thank you cbagwell for your answer, i could i've been stuck for long because of the safety feature.

closed 2023-03-18 by github-actions[bot]