MarlinFirmware/Marlin #24962: [BUG] MPC autotune ignores input arguments

gudvinr, 2022-11-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When you run `M306 T` with additional parameters, `M306` parser ignores all of them and uses shortcut to go straight to `MPC_autotune`.

MPC autotune doesn't tune heater power and filament heat capacity so it is not unreasonable to assume that you'd want to re-tune CRAF parameters when you change heater or when you want to tune MPC for new filament, for example.

So, to configure MPC parameters you need to run `M306 Pxx Hxx` first and only then use `M306 T`.  
That leaves user in a situation where you have inconsistent MPC parameters: before you run `M306 T` you'll have CRAF values for old setup and PH parameters for new one.  
Reliance on human interaction to responsibly run one command after another is quite obviously a bad way of preventing user errors.

This also means that if command is halted, these invalid parameters could be left as is without resetting them.

Same could be said for `E` but I don't have multiple extruders so I am not sure how to proceed here. Ideally, firmware should select extruder passed in `E` parameter and then start autotune but that's out of scope of what I care about.

### Bug Timeline

When MPCTEMP was introduced

### Expected behavior

When user runs `M306 T [Pxx] [Hyy]` with P and H optionally passed, autotune routine should take into account these parameters instead of using current ones.

If for some reason autotune procedure doesn't finish successfully, P and H should not be set as current ones.

On autotune finish new values of P and H should be set along with CRAF

### Actual behavior

P and H ignored during autotune

### Steps to Reproduce

1. Run `M306 Hxx T` where xx is not the same as current H value
2. Wait for autotune to finish
3. Run `M306` to see that `H` value was not changed

### Version of Marlin Firmware

2.1.1

### Printer model

Artillery Genius Pro

### Electronics

Stock

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9941603/Configuration.zip)

tombrazier, 2022-11-14:
One potential downside to redefining the behaviour of `M306 T` is this way is it removes the possibility of `M306 T Pxx` being a way of telling Marlin to tune MPC with something other than the maximum power. This would be a helpful feature for high powered hotends that heat up really quickly and would benefit from tuning at a lower power setting.

gudvinr, 2022-11-14:
Why do you think it removes this? You mean that way P will be used as max power setting and tune power?

Right now I don't see if P used as anything in tune procedure though.

tombrazier, 2022-11-15:
Yes, I meat that P is the obvious parameter to use in conjunction with T to specify power whilst testing.

gudvinr, 2022-11-15:
But it is not used now, right? Or am I missing something?

tombrazier, 2022-11-15:
No it's not used now. I'd just like to use it that way when I get the chance to implement the power limiting option one day. Or for someone else to implement it. It is needed by people with fast heating hotends.

gudvinr, 2022-11-15:
I'd suggest renaming P to M (as in **M**ax Power) and either use P for your intended purpose or call it N (as in **N**ominal Power).

But I am worried that current users won't be happy about that, however MPC marked as experimental now so I am not sure how things are working here.

Separate power values kinda make sense because using one variable for power values that have different semantics might be a bit confusing anyway.

Another option I can think of is to pass power as an argument of T, like Txx. If xx is omitted, then it behaves like it does now.  
It is backwards compatible, solves your issue but kinda clunky.

But all this looks like bikeshedding though, what do you think about issue itself?

thinkyhead, 2022-11-24:
> I'd suggest renaming P to M (as in Max Power)

No parameters can be named M unfortunately.

gudvinr, 2023-01-24:
Yes, I kinda think it still has a value if you don't have non-UART display that has no marlin UI

closed 2023-04-06 by github-actions[bot]