MarlinFirmware/Marlin #24963: [BUG] EEPROM error with M502 then M500

ellensp, 2022-11-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With debugging addded to see what is going on

Send M502
> echo:Hardcoded Default Settings Loaded
> ok

No problems so far

Send: M500
> echo:eeprom_error: 0          (my added debugging) 
> echo:eeprom_index: 607     (my added debugging
> echo:offset: 603                   (my added debugging
> Error:Field motor_current_setting mismatch.
> ok

So position in eeprom is out by 4 

 

### Bug Timeline

4 weeks

### Expected behavior

This should not error

### Actual behavior

It errors!

### Steps to Reproduce

1. Compile firmware with provided Configs and upload
2. execute M502 them M500
3. pull your hair out

### Version of Marlin Firmware

Bugfix-2.1.x

### Printer model

not applicable

### Electronics

mks tinybee

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9943029/Configuration.zip)

Debugging I added
```DIFF
diff --git a/Marlin/src/module/settings.cpp b/Marlin/src/module/settings.cpp
index f76b5afe74..c20761ac1a 100644
--- a/Marlin/src/module/settings.cpp
+++ b/Marlin/src/module/settings.cpp
@@ -1429,6 +1429,12 @@ void MarlinSettings::postprocess() {
     // Motor Current PWM
     //
     {
+      //EEPROM_ASSERT( eeprom_error || eeprom_index == offsetof(SettingsData, motor_current_setting) + EEPROM_OFFSET, "Field " STRINGIFY(motor_current_setting) " 
+
+      SERIAL_ECHO_MSG("eeprom_error: ",eeprom_error);
+      SERIAL_ECHO_MSG("eeprom_index: ",eeprom_index);
+      SERIAL_ECHO_MSG("offset: ",      offsetof(SettingsData, motor_current_setting) + EEPROM_OFFSET);
+
       _FIELD_TEST(motor_current_setting);
 
       #if HAS_MOTOR_CURRENT_SPI || HAS_MOTOR_CURRENT_PWM
```

 ### Note this issue is not with motor_current_setting  but with planner_extruder_advance_K

we have
```cpp
  //
  // LIN_ADVANCE
  //
  float planner_extruder_advance_K[DISTINCT_E]; // M900 K  planner.extruder_advance_K
``` 
where DISTINCT_E = 1

later we have 
```cpp
    //
    // Linear Advance
    //
      _FIELD_TEST(planner_extruder_advance_K);

      #if ENABLED(LIN_ADVANCE)
        EEPROM_WRITE(planner.extruder_advance_K);
      #else
        dummyf = 0;
        for (uint8_t q = _MAX(EXTRUDERS, 1); q--;) EEPROM_WRITE(dummyf);
      #endif
    }
```
where the else is run but _MAX(EXTRUDERS, 1  returns 2
These are in conflict

The first block of code was changed 4 weeks ago in https://github.com/MarlinFirmware/Marlin/pull/24821/files

where
`float planner_extruder_advance_K[_MAX(EXTRUDERS, 1)]; // M900 K planner.extruder_advance_K`
became
`float planner_extruder_advance_K[DISTINCT_E]; // M900 K planner.extruder_advance_K`

I suspect the fix is as simple as changing 
`        for (uint8_t q = _MAX(EXTRUDERS, 1); q--;) EEPROM_WRITE(dummyf);`
to
`        for (uint8_t q = DISTINCT_E); q--;) EEPROM_WRITE(dummyf);`

This compiles. but I don't have time to do exhaustive testing right now.

GMagician, 2022-11-05:
why not use COUNT(planner_extruder_advance_K) (sizeof(planner_extruder_advance_K)/sizeof(planner_extruder_advance_K[0])

valterfc, 2022-11-07:
I have a similar bug.

When active "EXTRUDERS=2" and "SINGLENOZZLE", the error "EEPROM DATASIZE ERROR" occurs, testing with M502, after M500.

If "EXTRUDERS=1" and "SINGLENOZZLE" off, the error does not occur.

Board: BIGTREE_OCTOPUS_PRO_V1_F429_USB
Marlin Version: Bugfix-2.1.x.

thinkyhead, 2022-11-07:
> I have a similar bug.

If the committed patch doesn't help, you can narrow down the cause by doing a build with `#define DEBUG_EEPROM_READWRITE` added to your config. This will add extra debugging to emit the name of the EEPROM field following the one that is causing the size problem.

commit 885e923b (2022-11-07): 🩹 Fix EEPROM write for !LIN_ADVANCE

Fixes #24963
Followup to #24821
closed 2022-11-07 by thinkyhead