MarlinFirmware/Marlin #24968: [BUG] Motors make louder and grinding sound after homing

classicrocker883, 2022-11-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I think this is particular to the Generic STM32F103RE environment, and/or Voxelab Aquila boards.

After having the BLtouch and UBL enabled, the motors would be fine when moving manually in the Move Axis submenu, but once it goes to Home, all at once or any which axis, the motor energizes and moves as if it is out of step. it makes a louder and grinding type of noise and more vibration. 

I've only had it happen with the Maple env. and ive been able to get it to come and go, but it is unclear how or what settings were changed. I'll do some more testing and come back with anymore information.

but I since think it is related to the BL touch.

### Version of Marlin Firmware

bugfix-2.1.x most current

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Nikolay-Po, 2022-12-15:
Hi! One man reported the same bug - grinding at endstops with BTT SKR 1.4 turbo (Cortex M3 MCU core).
He said, both X and Y endstops triggering with a lag comparable to one second duration. Z endstop seems to working OK.

Nikolay-Po, 2022-12-15:
Here are configuration files which are exhibiting an issue.

ilko-k, 2023-01-08:
Hi, I got a similar issue when I launched a new print after having flashed the v2.1.2.

In my start gcode, Z axis is raised before the print really starts. In the this version, the feedrate seems to be wrongly calculated, or it has something to do with the new TMCs code structure maybe… The motor is louder, suddenly gets overboosted and jumping steps while grinding at 3000+ feedrate. It's a G0 move and I defined variable G0 feedrates.

This does not occur with v2.1.1.

ProtomakerSprint, 2023-03-21:
grinding sound, I've had this because, (I know it doesn't make sense), the Printer bed is not level

classicrocker883, 2023-03-23:
look at **/Marlin/src/module/stepper.h**

the cycles values have been changed and may be the cause of the issue.

ilko-k, 2023-04-30:
hello, I bought a new printer, this time it's a coreXY with a MKS Robin Nano board. Flashed v2.1.2 and rolled back to 2.1.1 after less than one minute cause that grinding issue is still there. Maybe it has something to do with TMC drivers (still) or MKS brand in general ?

Reminder : The issue also occurs on my Tevo Flash (MKS GEN L, TMC drivers)

classicrocker883, 2024-09-14:
if anyone has this issue (generally noticed using Maple env boards) one fix is to add `#define OLD_ADAPTIVE_MULTISTEPPING` to **Configuration_adv.h**

closed 2024-12-29 by github-actions[bot]