MarlinFirmware/Marlin #24969: MKS ROBIN MNI V2.0  MKS ROBIN TFT24 V1  Po wgraniu  Robin_mini.bin   Pojawia się napis "Complete"  brak restartu

qulet, 2022-11-08:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

MKS ROBIN MNI V2.0  MKS ROBIN TFT24 V1  Po wgraniu  Robin_mini.bin   Pojawia się napis "Complete"  brak restartu

### Bug Timeline

stary problem

### Expected behavior

Restart oprogramowanie  ekran Marlina z przyciskami sterowania

### Actual behavior

 Płyta nie restartuje LCD wyświetla napis "Complete"

### Steps to Reproduce

Merlin V2,1 nie jestem w stanie z kompilować  kod dla MKS Robin Mini

### Version of Marlin Firmware

Marlin-2.0.6.1 

### Printer model

Banach School

### Electronics

MKS Robin Mini V1

### Add-ons

LCD MKS ROBIN TFT24 V1.0

### Bed Leveling

No Bed Leveling

### Your Slicer

Simplify3D

### Host Software

Same as my slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/9961330/Marlin.zip)

ellensp, 2022-11-08:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2022-11-08 by ellensp

qulet, 2022-11-08:
Nie mogę dokonać kompilacji:
Wykonywanie zadania: C:\Users\Marek\.platformio\penv\Scripts\platformio.exe run --environment mks_robin_mini 

Processing mks_robin_mini (board: genericSTM32F103VE; platform: ststm32@~12.1; framework: arduino)
----------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
Error: Build environment 'mks_robin_mini' is incompatible
with BOARD_RAMPS_14_EFB. Use one of these: mega2560, mega1280
=============== [FAILED] Took 1.56 seconds ===============

Environment     Status    Duration
--------------  --------  ------------
mks_robin_mini  FAILED    00:00:01.565
========= 1 failed, 0 succeeded in 00:00:01.565 =========

 *  Działanie procesu terminalu „C:\Users\Marek\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'mks_robin_mini'” zostało zakończone z kodem zakończenia: 1. 
 *  Terminal zostanie ponownie użyty przez zadania. Naciśnij dowolny klawisz, aby go zamknąć. 

 *  Wykonywanie zadania: C:\Users\Marek\.platformio\penv\Scripts\platformio.exe run --environment mks_robin_mini 

Processing mks_robin_mini (board: genericSTM32F103VE; platform: ststm32@~12.1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:38,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
e:\mini\marlin-bugfix-2.1.x\marlin-bugfix-2.1.x\marlin\src\core\boards.h:480:64: error: operator '==' has no right operand
  480 | #define _MB_1(B)  (defined(BOARD_##B) && MOTHERBOARD==BOARD_##B)
      |                                                                ^
e:\mini\marlin-bugfix-2.1.x\marlin-bugfix-2.1.x\marlin\src\core\macros.h:182:29: note: in expansion of macro '_MB_1'
  182 | #define _DO_1(W,C,A)       (_##W##_1(A))
      |                             ^
e:\mini\marlin-bugfix-2.1.x\marlin-bugfix-2.1.x\marlin\src\core\macros.h:222:28: note: in expansion of macro '_DO_1'
  222 | #define __DO_N(W,C,N,V...) _DO_##N(W,C,V)
      |                            ^~~~
e:\mini\marlin-bugfix-2.1.x\marlin-bugfix-2.1.x\marlin\src\core\macros.h:223:28: note: in expansion of macro '__DO_N'
  223 | #define _DO_N(W,C,N,V...)  __DO_N(W,C,N,V)
      |                            ^~~~~~
e:\mini\marlin-bugfix-2.1.x\marlin-bugfix-2.1.x\marlin\src\core\macros.h:224:29: note: in expansion of macro '_DO_N'
  224 | #define DO(W,C,V...)       (_DO_N(W,C,NUM_ARGS(V),V))
      |                             ^~~~~
e:\mini\marlin-bugfix-2.1.x\marlin-bugfix-2.1.x\marlin\src\core\boards.h:481:19: note: in expansion of macro 'DO'
  481 | #define MB(V...)  DO(MB,||,V)
      |                   ^~
e:\mini\marlin-bugfix-2.1.x\marlin-bugfix-2.1.x\marlin\src\pins\pins.h:514:7: note: in expansion of macro 'MB'
  514 | #elif MB(MKS_ROBIN_MINI)
      |       ^~
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
e:\mini\marlin-bugfix-2.1.x\marlin-bugfix-2.1.x\marlin\src\pins\pins.h:881:6: error: #error "MOTHERBOARD not defined! Use '#define MOTHERBOARD BOARD_...' in Configuration.h."
  881 |     #error "MOTHERBOARD not defined! Use '#define MOTHERBOARD BOARD_...' in Configuration.h."
      |      ^~~~~
Error: Failed to parse Marlin features. See previous error messages.
============================================================================== [FAILED] Took 1.27 seconds ==============================================================================

Environment     Status    Duration
--------------  --------  ------------
mks_robin_mini  FAILED    00:00:01.271
========================================================================= 1 failed, 0 succeeded in 00:00:01.271 =========================================================================

 *  Działanie procesu terminalu „C:\Users\Marek\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'mks_robin_mini'” zostało zakończone z kodem zakończenia: 1. 
 *  Terminal zostanie ponownie użyty przez zadania. Naciśnij dowolny klawisz, aby go zamknąć.