MarlinFirmware/Marlin #24979: Steps skipping or sudden current variation while traveling

Andre1902, 2022-11-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

i can see on long travels some random steps skipping (X and Y). in the last few weeks i tried with changing every possible setting  i know in the firmware but i cant figure out the problem (such as acceleration, speed, current, sensorless homing, silent stepper driver, interpolation...). i sliced with other slicer and with cura i could manage to get some better results, maybe due to the acceleration and jerk control, but with prusa slicer i couldnt find a solution to the issue. i also tried to change some marlin version from 2.x up to the last bugfix.

### Expected behavior

after motherboard ubgrate from Creality 4.2.2 to BTT SKR Mini E3 V3.0

### Version of Marlin Firmware

Marlin-bugfix-2.1.x

### Printer model

Ender 3

### Electronics

BTT SKR Mini E3 V3.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/9995475/config.zip)

oliver-eifler, 2022-11-18:
You should deactivate S_CURVE_ACCELERATION - see https://github.com/MarlinFirmware/Marlin/pull/24955 for explanation

and set the currents according to your motors

Andre1902, 2022-11-18:
Ive also tryed to disable it, current was set to 800 according to my TMC2209, a little bit higher than marlin default for ender 3. I solved my issue yesterday by reading this #19217. Deactivated SEGMENT_LEVELED_MOVES and now it works fine, no more stuttering.

Fredrossi046, 2023-01-05:
Problem is not possible to be solved for me, because i have the pre-compiled firmware. 
I installed this firmware : firmware_Ender6_BLT_50W_ET_DW7.4.6.bin.zip  from  insanityautomation.com/index.php/firmware-projects

Machine was working fine with small prints, but now i get the stuttering problem as well, at least with larger prints. Disable ABL is all i can do, but that won't solve it.  Deactivating SEGMENT_LEVELED_MOVES is not possible for me, as i can not open the bin-file in Arduino-IDE. Or is there a way i can open, alter and place it back on SD for updating firmware?

I changed current for the stepper driver (at mainboard), but that is all i can do i guess. Strange thing is, i printed a feedrate test, that is larger than the bed level points of my 25 ABL points, but that did not have any problems. 

Mainboard is stock, just the above update with initial no problems last few days.

Andre1902, 2023-01-06:
i think this is what you are looking for: [https://github.com/bigtreetech/Marlin/tree/SKR-mini-E3-V3.0-G0B1](url)

ellensp, 2023-01-06:
Neither the bigtreetech or the insanity automation forks are supported here.
If your not running stock Marlin code from https://github.com/MarlinFirmware/Marlin, you are in the wrong place.

closed 2024-09-24 by github-actions[bot]