MarlinFirmware/Marlin #24981: [BUG] Changes in Marlin Flow Rate from Printer LCD Tune Command are Not Reflected in Status Screen

helakejr, 2022-11-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Goal: Adjust Flow Rate by using the Marlin Tune command to adjust current flow rate, from printer LCD.
BUG: Marlin Status Screen for print still shows 100% although I changed to 101%, 104%, ...
Note: 
1) Re-entering Tune->Flow Rate confirms flow rate is set the modified value.
2) Doesn't matter if flow rate is saved to EEPROM or not.
3) Did the flow rate change or not?

Machine: Anet A8
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/9995600/Configuration.zip)

MOTHERBOARD: BOARD_ANET_10
Marlin Version: Marlin 2.1.1, Marlin Bugfix 2.1.x
Build: Visual Studio Code with Platform IO Plugin

### Bug Timeline

Marlin 2.1.1

### Expected behavior

Marlin Status Screen for print should show changed Flow Rate, agreeing with set Tune->Flowrate.

### Actual behavior

Marlin Status Screen for print shows 100% for any value set for Flowrate.

### Steps to Reproduce

1. Start any print
2. Modify flow rate to any value by using the Tune->Flowrate command from the printer LCD
3. Display Print Status Screen and observe Flow Rate still at 100%
4. 

### Version of Marlin Firmware

Marlin Bugfix 2.1.x

### Printer model

Anet A8

### Electronics

Stock

### Add-ons

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

jmz52, 2022-11-12:
F/R on status screens stands for Feed Rate, not Flow Rate.
These are two different parameters.

helakejr, 2022-11-12:
My apologies for my misunderstanding. Thank you

closed 2022-11-12 by ellensp