MarlinFirmware/Marlin #24988: [BUG] BTT MINI12864 Wrong NeoPixel Colours (Can't select anything other than NEO_RGB)

CNCNoob, 2022-11-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi there!

Have a Bigtreetech MINI12864 LCD screen attached to BTT SKR V3.0. 
Everything works fine other than the NeoPixels which seem to be the wrong colours, I would like to change `NeoPixel_Type  NEO_RGB` to `NeoPixel_Type  NEO_GRBW` as Specified on the website of the LCD display: https://wiki.fysetc.com/Mini12864_Panel/ But Marlin throws an error "#error "Your FYSETC/MKS/BTT Mini Panel requires NEOPIXEL_TYPE to be NEO_RGB.""

Why is this error included in the "Sanity Check" document when it seems to be contrary to the manufacturer's website instructions? Can I just overwrite this sanity check so it doesn't throw the error or will that bust the rest of the code?
I've attached the configs here: [Configs.zip](https://github.com/MarlinFirmware/Marlin/files/10001931/Configs.zip)

cheers

### Bug Timeline

New bug

### Expected behavior

I expect the correct colours

### Actual behavior

Wrong colours due to incorrect NeoPixel choice.

### Steps to Reproduce

1. Flash
2. Turn on
3. Check colours

### Version of Marlin Firmware

Marlin 2.1.1

### Printer model

Home made :)

### Electronics

Upgrade board: BTT SKR V3.0

### Add-ons

BTT Mini12864 V1.0 LCD

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/10001946/Configs.zip)

ellensp, 2022-11-14:
The error is correct, the page you are referring to is wrong.

On the back of the lcd are 3 chips labeled ws2811 these are the rgb pixel drivers
Here is the data sheet https://cdn-shop.adafruit.com/datasheets/WS2811.pdf

These are "Signal line 256 Gray level 3 channal Constant current LED drive IC" 

The 3 channels being Red Green and Blue. there is no white channel

CNCNoob, 2022-11-14:
Hey there thanks for your reply! 

I think the LCD I have (it says on it BTT MINI12864) Must be different from the one your describing..

As in the website I had before it has 3 LED's but they are positioned as shown below. after some code experimentation, the following seemed to correct the colour pins, but they seem to have alot less than 256 steps (altho I don't know if this is editable in the code)
`NeoPixel_Type NEO_BRG`

[image]

ellensp, 2022-11-14:
Each of those 3 leds is really 3 leds, you have 9 leds total (setup as 3 rgb neopixels) 
Those leds you have circled is really a red and green and a blue led in on package

If you enable just one channel (red greeen or blue) and take a close look at the led  you will note that it glows from the left, or center or right in a different base colour

CNCNoob, 2022-11-14:
Yes I understand that thankyou,

Is there a way to reduce the resolution of the neopixel library? Since I have the colours the correct way now (BRG) but there is only 32 steps not 256.

ellensp, 2022-11-14:
I have no idea what your talking about
The LCD menu lets you set custom colours, and gives you the option of 0-255
The gcode command M150 also takes values from 0 255

CNCNoob, 2022-11-14:
Yeah ok, seems like the NeoPixels on my board are not adafruit ones then 😢 
I'll upload a video so you can see what's happening.

CNCNoob, 2022-11-14:
https://user-images.githubusercontent.com/52121166/201642016-2e33fef3-596e-4658-b19b-fde71496e5e7.mp4

ellensp, 2022-11-14:
That is because you using the wrong nepixel type It is and must be a set to NEO_RGB
You trying to correct a timing issue with the wrong neopixel type, but then you change the value to far it break the timing again

thisiskeithb, 2022-11-14:
`NEO_RGB` is correct for this LCD and I’ve confirmed this using a real BTT Mini 12864 V1 on an SKR V3 and multiple motherboards.

Don’t bypass those sanity checks.

closed 2022-11-14 by thisiskeithb