MarlinFirmware/Marlin #24990: [BUG] Issue with SPI_ENDSTOPS

rondlh, 2022-11-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using SPI_ENDSTOP the endstop status is stored in a variable (live_state in endstops.cpp) which keeps track of the endstop status. The status needs to be reset when homing otherwise an old state will be used, which probably shows a trigger of the endstops (clear_endstop_state() in endstops.cpp).
This status is reset when starting a homing cycle, but not before the 2nd homing run is executed when doing a homing "bump". This becomes obvious when setting a large (5-10mm) homing bump (HOMING_BUMP_MM). In that case you can see that the carriage moves away from the endstop after the first homing. Next the second homing is execute, but the carriage never moves towards the endstop because the endstop status is not cleared, so the endstop seems to be triggered immediately.

### Bug Timeline

Don't know

### Expected behavior

Execute the homing after the homing bump correctly

### Actual behavior

The homing after the homing bump is not working because the endstop status was not cleared from the previous homing.

### Steps to Reproduce

1. Enable SPI_ENDSTOP and set HOMING_BUMP_MM to a large number (10, 10, 0)
2. Execute a homing and monitor the carriage movements after the first homing.

My configuration is attached below.
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/10009224/Configurations.zip)

### Version of Marlin Firmware

2.1.x bugfix (November 15th 2022)

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### Add-ons

BTT TFT35 V1.0

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thinkyhead, 2022-11-24:
It might fix it to make this change in `motion.cpp` inside of `homeaxis`:
```diff
      do_homing_move(axis, -bump, TERN(HOMING_Z_WITH_PROBE, (axis == Z_AXIS ? z_probe_fast_mm_s : 0), 0), false);

+     // Clear endstop state for polled stallGuard endstops
+     TERN_(SPI_ENDSTOPS, endstops.clear_endstop_state());

      #if ENABLED(DETECT_BROKEN_ENDSTOP)
```
If that does the trick, then I'll merge this patch asap.

rondlh, 2022-11-29:
> It might fix it to make this change in `motion.cpp` inside of `homeaxis`:
>
> ```diff
>       do_homing_move(axis, -bump, TERN(HOMING_Z_WITH_PROBE, (axis == Z_AXIS ? z_probe_fast_mm_s : 0), 0), false); [...]

That's exactly how I solved this issue, great job!

closed 2023-02-08 by github-actions[bot]