MarlinFirmware/Marlin #24997: [BUG] bugfix-2.1.x INPUT_SHAPING BTT SKR 3 Touch Probe Ignored

spikked27, 2022-11-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With INPUT_SHAPING enabled homing moves are extremely slow and z-probe is ignored causing the head to crash into the bed.

### Bug Timeline

New with bugfix-2.1.x

### Expected behavior

Expected homing moves to remain the same with or without INPUT_SHAPING and for the z-probe to work as expected

### Actual behavior

Homing moves crawl and head crashes into bed

### Steps to Reproduce

1. Take a working bugfix-2.1.x configuration
2. Enable INPUT_SHAPING
3. Compile and load firmware
4. Initiate home all axis

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 3 V2

### Electronics

BTT SKR 3

### Add-ons

CR Touch Bed Probe

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/10043053/config.zip)

thisiskeithb, 2022-11-18:
Input shaping is still being worked on:

- https://github.com/MarlinFirmware/Marlin/pull/24951

tombrazier, 2022-11-18:
I think this is likely a duplicate of #24927, which is resolved in #24951.

thisiskeithb, 2022-11-18:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/24927

closed 2022-11-18 by thisiskeithb