MarlinFirmware/Marlin #24998: Memory Leak after Abort?

inspectionsbybob, 2022-11-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

After an aborted print, upon starting a NEW print one or more axis "creeps" during warm up.  Typically, it is the X-Axis (or one of them in an IDEX config).  

It is noticed during the Bed/extruder warm up BEFORE a G28.

(I warm the bed before doing any Bed-Leveling or Homing)

If allowed to proceed, it causes mispositioned extruders and other random issues.

My Startup Gcode is the following

G91                                                                   ; relative moves until we know where we are
G1 Z5 F5000                                                     ; lift nozzle
M104 S200                                                      ; start heating Nozzle
M117 Waiting On Bed Temp
M190 S[first_layer_bed_temperature]            ; set bed temp AND WAIT for BED temp to be reached

G28 ; home all axes after Bed to Temp
M420 S1                                                         ; turn bed leveling back on after G28
T0                                                                  ; left extruder 
G90                                                               ; absolute positioning 
M117 Waiting on Extruder Temp
.
.
.

Sometimes it takes more than one abort in a row for it to show...

### Bug Timeline

This version 2.1.1

### Expected behavior

No Movement 

### Actual behavior

Right extruder (Mine is IDEX) E1/T1 slowly creeps left as the system warms up before a G28

### Steps to Reproduce

1.  Start Print on MULTI-EXTRUDER printer that warms up before G28
  (Noticed while only using ONE extruder E0/t0)
2. ABORT print
3. Start Print
4. Abort Print
5. Start print and watch T1 as the system warms up before G28

### Version of Marlin Firmware

2.1.1

### Printer model

JGMaker Artist

### Electronics

Robin Pro

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Slicer is SUPERSLICER based on Prusa

[2.1.1 Configs.zip](https://github.com/MarlinFirmware/Marlin/files/10043117/2.1.1.Configs.zip)

thinkyhead, 2022-11-24:
I suspect there is some kind of PWM noise getting into the driver from a nearby source, because there isn't any immediately obvious mechanism in Marlin for a very slow short move to be inserted at the end of a print abort. Of course, after a print abort, re-homing is indicated, since abort is known to cause loss of position due to the sudden stopping. But anyway, more experiments are needed to demonstrate when this does occur versus when it does not occur, to see if some causes can be isolated, or at least eliminated.

Can you try adding an `M18 XY` before the point where you see this axis creeping happening? That could give some clue whether a planner move is hanging around, since that would immediately re-enable the steppers. If that doesn't make any difference, then we can look into other potential issues.

inspectionsbybob, 2023-02-17:
problem still exist in latest full version. Haven't tried any bugfix versions.
Very anoying as if it hits end-stop it doesn't stop, it just grinds.
Always very slow movement, maybe 5mm/sec. 

Display typically shows bed warming up...

Again, it ONLY happens after an aborted print...

still looking for a "reset" I can put somewhere to clean stack and memory after an aborted print...

closed 2023-05-01 by github-actions[bot]