MarlinFirmware/Marlin #25008: [BUG] Polargraph feature doesn't seem to do math correctly with smaller sized polargraphs

MacGyverr, 2022-11-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using the following settings:

#define POLARGRAPH_MAX_BELT_LEN 673.0
#define X_BED_SIZE 454.0
#define Y_BED_SIZE 634.0
#define X_MIN_POS (-X_BED_SIZE/2)
#define Y_MIN_POS (-Y_BED_SIZE/2)
#define X_MAX_POS (X_BED_SIZE/2)
#define Y_MAX_POS (Y_BED_SIZE/2)
#define BED_CENTER_AT_0_0
#define MANUAL_X_HOME_POS 0
#define MANUAL_Y_HOME_POS (Y_MAX_POS - ( sqrt( ( sq(POLARGRAPH_MAX_BELT_LEN) ) - (sq(X_BED_SIZE/2)) ) ) )

the following lines will not work in order. (you can't move vertically 584mm down)
G1 X199.500 Y282.150 F3000.0
G1 X199.500 Y-282.150 F3000.0

### Bug Timeline

I can't compile 2.0 for polargraph so I don't know

### Expected behavior

I expect this to move left then right, then down, then left, then up.

G1 X-199.500 Y282.150 F3000.0
G1 X199.500 Y282.150 F3000.0
G1 X199.500 Y-282.150 F3000.0
G1 X-199.500 Y-282.150 F3000.0
G1 X-199.500 Y282.150 F3000.0

### Actual behavior

With my correct dimensions installed it ignores the vertical movements at those distances so it just goes left, then right, then left and stops. (smaller vertical movements work)

It should be able to go from -317 to +317 (634mm) and the movement length in those commands is only 584mm

### Steps to Reproduce

#define POLARGRAPH_MAX_BELT_LEN 673.0
#define X_BED_SIZE 454.0
#define Y_BED_SIZE 634.0
#define X_MIN_POS (-X_BED_SIZE/2)
#define Y_MIN_POS (-Y_BED_SIZE/2)
#define X_MAX_POS (X_BED_SIZE/2)
#define Y_MAX_POS (Y_BED_SIZE/2)
#define BED_CENTER_AT_0_0
#define MANUAL_X_HOME_POS 0
#define MANUAL_Y_HOME_POS (Y_MAX_POS - ( sqrt( ( sq(POLARGRAPH_MAX_BELT_LEN) ) - (sq(X_BED_SIZE/2)) ) ) )

Compile.
execute M502

the following lines will not work in order. (you can't move vertically 584mm down)
G1 X199.500 Y282.150 F3000.0
G1 X199.500 Y-282.150 F3000.0

strange thing is that if eeprom has 
POLARGRAPH_MAX_BELT_LEN 1000.0
Y_BED_SIZE 1000.0
stored incorrectly in it but with the smaller sizes compiled into the firmware, it will work correctly.

It's like something uses the settings in eeprom to decide if the movement is valid (and doesn't work with dimensions as small as mine), but something else uses the compiled numbers to calculate and move.

If you move manually to the middle you can manually move Y all the way from -317 to +317,  it almost like it uses eeprom settings to create the "bed" and at smaller sizes it's creating a bed that is circular in shape rather than square so it can't draw corners.

### Version of Marlin Firmware

bugfix-2.1.x Aug 6

### Printer model

polargraph

### Electronics

Mega 2560, Ramps 1.4

### Your Slicer

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

gcode generated with Makelangelo 7.32.0
[Marlin-configs.zip](https://github.com/MarlinFirmware/Marlin/files/10072766/Marlin-configs.zip)

MacGyverr, 2022-11-23:
Here is how the numbers relate to my hardware (the larger ones are the squares for the math).
[image]

thinkyhead, 2022-11-24:
Thanks for the report! It sounds like a good question for @i-make-robots to review, and if a correction is due then I'm sure it can be patched pretty quickly.

i-make-robots, 2022-11-24:
Cartesian limits are in motion.cpp ~ 925
```c
        LIMIT(target.x, draw_area_min.x, draw_area_max.x);
        LIMIT(target.y, draw_area_min.y, draw_area_max.y);
```
and polargraph limits ~340
```c
      const float d1 = rx - (draw_area_min.x),
                  d2 = (draw_area_max.x) - rx,
                   y = ry - (draw_area_max.y),
                   a = HYPOT(d1, y),
                   b = HYPOT(d2, y);

      can_reach = (
           a < polargraph_max_belt_len + 1
        && b < polargraph_max_belt_len + 1
      );
```
looking at your numbers,

raw.x = 199.5
raw.y = -282.15
d1 = 516.5
d2 = -117.5
y = -599.15
hypot(d1,y) = 791.0454933 // <-- longer than your belt, will fail
hypot(d2,y) = 610.562826  // acceptable

your belt must be at least as long as `hypot(X_BED_SIZE,Y_BED_SIZE)` if you want to reach every part of the bed.  That would be... 779.7897152.  Another option is to make smaller drawings in the same work area so that you never reach into the bottom corners and exceed the belt length.

MacGyverr, 2022-11-25:
I see now where I screwed up my math (been a long time since I used the Pythagorean theorem), and the fact the Makelangelo assumes you have at least a 1m belt, so it will output gcode that goes outside the possible area of the math I did with my belt length.
I actually have a large belt, but was having trouble moving the printing area lower on my whiteboard, likely because of the variables I was using instead of the actual numbers, that's why I had the belt set so small. 

Thanks for showing me where I went wrong. And thank you for all the work you put into the code. :) (especially the M665 stuff, that is going to be super helpful when I want to move my polargraph to different board sizes)

closed 2022-11-25 by MacGyverr