MarlinFirmware/Marlin #25009: [BUG] mmu2 does not reset with SW Reset

ChGeorgi, 2022-11-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin sends during boot ?X0 instead of X0
i fixed it by sending a "start" before X0 (but can also only be a blank)

### Expected behavior

reset of mmu2

### Actual behavior

mmu2 will be disabled

### Steps to Reproduce

connect mmu2 without reset cable

### Version of Marlin Firmware

2.0.9.2

### Printer model

self designed almost like prusa mk3s

### Electronics

skr 1.4 /skr 2.0

### Add-ons

MMU2S with clone board 

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

nothing else

thinkyhead, 2022-11-24:
The code is only sending "X0" so it's not clear where the "?" is coming from on your serial line.
```cpp
  MMU2_SERIAL.begin(MMU_BAUD);
  safe_delay(10);
  MMU2_COMMAND("X0"); // Send soft reset
```
You can try experimenting with a longer `safe_delay`, or you might try adding `MMU2_SERIAL.flushTX();` to see if it makes a difference in the reliability of startup.

ChGeorgi, 2022-11-24:
The '?' spears directly at startup. Followed by the X0 later. So Its something before.

GMagician, 2022-11-25:
May this be caused by noise on usart line? You don't send anything but noise causes bits to be loaded into shift register.

closed 2024-12-21 by github-actions[bot]