MarlinFirmware/Marlin #25010: [BUG] (bug summary) Z_SAFE_HOMING is broken

francuz152, 2022-11-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

while homing Z axis the printer "sets" the X and Y value to the correct for z safe homing (middle of the bed) but the extruder doesn't actually moves,
if i home x and y, then manually move to the center of the bed and then home Z, it goes right down and probe correctly but any command that have a homing in it is impossible to use (like g29)

### Expected behavior

extruder actually moves before homing Z

### Actual behavior

extruder after homing X and Y go strait down (in my setup crash into the bed) 

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin 2.1.1 (Nov 23 2022 11:46:42)

### Printer model

anet A8 plus (moded)

### Electronics

SKR 1.7 Turbo

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

config : https://pastebin.com/zpQ0yLNc
config adv : https://pastebin.com/VK9Gf5Ly
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10075753/Marlin.zip)

ellensp, 2022-11-23:
reset your eeprom to firmware defaults
M502 then M500, you should do this after every upload until firmware is in it final configuration
or enable EEPROM_INIT_NOW so that it sets eeprom to defaults on every upload of new firmware

francuz152, 2022-11-23:
done it multiple times, no change

thisiskeithb, 2022-11-23:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun, Repetier Host, or OctoPrint.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

francuz152, 2022-11-23:
[https://pastebin.com/2LeVxrvm](url)

the issue is that when it is supposed to go to the middle of the bed it just don't move and go strait down
homing X and Y work fine
[logs.txt](https://github.com/MarlinFirmware/Marlin/files/10078551/logs.txt)

thinkyhead, 2022-11-24:
According to the log there is a move being sent to the planner after X and Y have homed to `X283` `Y-3`…
```
Recv: <<< do_blocking_move_to  X111.00 Y121.00 Z5.00
```
So it's a mystery why the commanded move is not actually taking place. It implies that the X and Y steppers have gone into standstill, or something. See if it makes any difference to set the acceleration really low with `M201 X500 Y500` or `M204 T500` before homing.

ellensp, 2022-11-24:
Prom the posted configs you have 

#define X_HOME_DIR 1
#define USE_XMIN_PLUG
#define USE_XMAX_PLUG

But what is plugged into USE_XMIN_PLUG? I suspect nothing.

I suspect this is why It will not move

Your performing a Z home with z_safe_homie marlin want to move toward the X min endstop from homing to max but the min endstpp is saying I am triggered don't move

Comment out #define USE_XMIN_PLUG

francuz152, 2022-11-24:
that done it, 
thank you very much

ellensp, 2022-11-24:
@thinkyhead do you want to chase this further?  add endstop triggered message to the debugging ?