MarlinFirmware/Marlin #25014: [BUG] Homing Probe not using Probe Delay

ravem0n, 2022-11-24:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Currently using a Piezo Orion and I set the Probe_Delay to 500. When I home and the hotend goes to the center, the moment it reaches the center it instantly starts probing. I lowered the sensitivity as low as it can get before it just doesn't trigger anymore. It would still once in a while get false positives. I tried setting the Probe_Delay to 2000 and it is still probing right away. When doing a probe repeatability test and when using UBL, the probe delay works. It just doesn't work on when I home the Z. Tried 2 boards and same behavior.

I upgraded from 2.0.7 and it was working fine with the old version. 

### Expected behavior

When Homing Z, I expect it to use the probe delay and wait 500ms

### Actual behavior

It starts probing right away.

### Version of Marlin Firmware

Marlin 2.1.1

### Printer model

Hypercube Evo

### Electronics

SKR Octopus V1.1; SKR Pro V1.2

### Add-ons

Piezo Orion 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_2_1_1_Conf.zip](https://github.com/MarlinFirmware/Marlin/files/10087568/Marlin_2_1_1_Conf.zip)

ellensp, 2022-11-25:
Probe_Delay is not in current marlin or in provided configs.

It would help if you used the actual key word DELAY_BEFORE_PROBING

slowbro, 2022-11-26:
Question: is your printer probing multiple times when homing? Does it pause on only the second/final probe?

ravem0n, 2022-11-26:
> Question: is your printer probing multiple times when homing? Does it pause on only the second/final probe?

Yes I have it probe twice. The pause does happen on the 2nd one. I have set it so it probes once and still the pause is not there.

closed 2023-02-07 by github-actions[bot]