MarlinFirmware/Marlin #25017: [BUG] enabling UBL results in error: thermalmanager was not defined in menu_ubl.cpp

mashuptwice, 2022-11-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When uncommenting `AUTO_BED_LEVELING_UBL`, compile fails with the following error:

```
Marlin/src/lcd/menu/menu_ubl.cpp:131:87: error: 'thermalManager' was not declared in this scope
     EDIT_ITEM(int3, MSG_UBL_HOTEND_TEMP_CUSTOM, &custom_hotend_temp, EXTRUDE_MINTEMP, thermalManager.hotend_max_target(0));
```

Is some #include missing in menu_ubl ?

### Version of Marlin Firmware

bugfix-2.1.x

### Electronics

MKS Tinybee

### Bed Leveling

UBL Bilinear mesh

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/10091839/conf.zip)

ellensp, 2022-11-25:
Provided config does not generate this error
you do haw ever get 
Marlin/src/inc/SanityCheck.h:2204:8: error: #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."
       #error "You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN."

Which is self explanatory 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

mashuptwice, 2022-11-25:
I don't get any error about the fan pin. Set correctly in conf_adv:
```
#define E0_AUTO_FAN_PIN 147
```

ellensp, 2022-11-25:
then you have disabled the sanity check.  

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/inc/SanityCheck.h#L2203-L2204

as you have
#define E0_AUTO_FAN_PIN 147
and
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/pins/esp32/pins_MKS_TINYBEE.h#L107
#define FAN_PIN                              147

these are clearly in conflict.

fixing this it compiles without issue

mashuptwice, 2022-11-25:
Just cloned the repo freshly to rule out any changes in my local copy, commented out FAN_PIN and  same error:
```
In file included from Marlin/src/lcd/menu/menu_ubl.cpp:31:0:
Marlin/src/lcd/menu/menu_ubl.cpp: In function 'void _lcd_ubl_custom_mesh()':
Marlin/src/lcd/menu/menu_ubl.cpp:131:87: error: 'thermalManager' was not declared in this scope
     EDIT_ITEM(int3, MSG_UBL_HOTEND_TEMP_CUSTOM, &custom_hotend_temp, EXTRUDE_MINTEMP, thermalManager.hotend_max_target(0));
                                                                                       ^
Marlin/src/lcd/menu/menu_item.h:277:39: note: in definition of macro '_MENU_INNER_F'
     MenuItem_##TYPE::action(flabel, ##V);                      \
                                       ^
Marlin/src/lcd/menu/menu_item.h:400:57: note: in expansion of macro '_MENU_ITEM_F'
 #define MENU_ITEM_F(TYPE, FLABEL, V...)                 _MENU_ITEM_F(TYPE, false, FLABEL, ##V)
                                                         ^
Marlin/src/lcd/menu/menu_item.h:446:58: note: in expansion of macro 'MENU_ITEM_F'
 #define EDIT_ITEM_F(TYPE, FLABEL, V...)                  MENU_ITEM_F(TYPE, FLABEL, ##V)
                                                          ^
Marlin/src/lcd/menu/menu_item.h:447:58: note: in expansion of macro 'EDIT_ITEM_F'
 #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_F(TYPE, GET_TEXT_F(LABEL), ##V)
                                                          ^
Marlin/src/lcd/menu/menu_ubl.cpp:131:5: note: in expansion of macro 'EDIT_ITEM'
     EDIT_ITEM(int3, MSG_UBL_HOTEND_TEMP_CUSTOM, &custom_hotend_temp, EXTRUDE_MINTEMP, thermalManager.hotend_max_target(0));
     ^
Marlin/src/lcd/menu/menu_ubl.cpp:131:87: error: 'thermalManager' was not declared in this scope
     EDIT_ITEM(int3, MSG_UBL_HOTEND_TEMP_CUSTOM, &custom_hotend_temp, EXTRUDE_MINTEMP, thermalManager.hotend_max_target(0));
                                                                                       ^
Marlin/src/lcd/menu/menu_item.h:282:58: note: in definition of macro '_MENU_INNER_F'
       (encoderLine == _thisItemNr, _lcdLineNr, flabel, ##V);   \
                                                          ^
Marlin/src/lcd/menu/menu_item.h:400:57: note: in expansion of macro '_MENU_ITEM_F'
 #define MENU_ITEM_F(TYPE, FLABEL, V...)                 _MENU_ITEM_F(TYPE, false, FLABEL, ##V)
                                                         ^
Marlin/src/lcd/menu/menu_item.h:446:58: note: in expansion of macro 'MENU_ITEM_F'
 #define EDIT_ITEM_F(TYPE, FLABEL, V...)                  MENU_ITEM_F(TYPE, FLABEL, ##V)
                                                          ^
Marlin/src/lcd/menu/menu_item.h:447:58: note: in expansion of macro 'EDIT_ITEM_F'
 #define EDIT_ITEM(TYPE, LABEL, V...)                     EDIT_ITEM_F(TYPE, GET_TEXT_F(LABEL), ##V)
                                                          ^
Marlin/src/lcd/menu/menu_ubl.cpp:131:5: note: in expansion of macro 'EDIT_ITEM'
     EDIT_ITEM(int3, MSG_UBL_HOTEND_TEMP_CUSTOM, &custom_hotend_temp, EXTRUDE_MINTEMP, thermalManager.hotend_max_target(0));
     ^
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mks_tinybee/src/src/lcd/menu/menu_ubl.cpp.o] Error 1
```

ellensp, 2022-11-26:
disabling the #define FAN_PIN 147 is the cause of your issue

ellensp, 2022-11-26:
Root cause:

Marlin/src/lcd/menu/menu_ubl.cpp
has #include "menu_item.h"

in menu_item.h is this code
```
#if HAS_FAN
  #include "../../module/temperature.h"
```

since the fan is disabled, temperature.h is not included 
and 
extern Temperature thermalManager;  is never set

so it cannot find thermalManager.

Quick fix
add this to top of Marlin/src/lcd/menu/menu_ubl.cpp after the other #includes
```
#if !HAS_FAN
  #include "../../module/temperature.h"
#endif 
```

ellensp, 2022-11-26:
minimal config needed to trigger this

```DIFF
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 87a98298a5..ae4a701457 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -88,6 +88,7 @@
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
   #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define FAN_PIN -1
 #endif
 
 /**
@@ -1316,7 +1317,7 @@
  * A Fix-Mounted Probe either doesn't deploy or needs manual deployment.
  *   (e.g., an inductive probe or a nozzle-based probe-switch.)
  */
-//#define FIX_MOUNTED_PROBE
+#define FIX_MOUNTED_PROBE
 
 /**
  * Use the nozzle as the probe, as with a conductive
@@ -1876,7 +1877,7 @@
 //#define AUTO_BED_LEVELING_3POINT
 //#define AUTO_BED_LEVELING_LINEAR
 //#define AUTO_BED_LEVELING_BILINEAR
-//#define AUTO_BED_LEVELING_UBL
+#define AUTO_BED_LEVELING_UBL
 //#define MESH_BED_LEVELING
 
 /**
@@ -2088,7 +2089,7 @@
  * - Allows Z homing only when XY positions are known and trusted.
  * - If stepper drivers sleep, XY homing may be required again before Z homing.
  */
-//#define Z_SAFE_HOMING
+#define Z_SAFE_HOMING
 
 #if ENABLED(Z_SAFE_HOMING)
   #define Z_SAFE_HOMING_X_POINT X_CENTER  // X point for Z homing
@@ -2174,7 +2175,7 @@
  *   M501 - Read settings from EEPROM. (i.e., Throw away unsaved changes)
  *   M502 - Revert settings to "factory" defaults. (Follow with M500 to init the EEPROM.)
  */
-//#define EEPROM_SETTINGS     // Persistent storage with M500 and M501
+#define EEPROM_SETTINGS     // Persistent storage with M500 and M501
 //#define DISABLE_M503        // Saves ~2700 bytes of flash. Disable for release!
 #define EEPROM_CHITCHAT       // Give feedback on EEPROM commands. Disable to save PROGMEM.
 #define EEPROM_BOOT_SILENT    // Keep M503 quiet and only give errors during first load
@@ -2730,7 +2731,7 @@
 // RepRapDiscount FULL GRAPHIC Smart Controller
 // https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
 //
-//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
+#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
 
 //
 // K.3D Full Graphic Smart Controller

```

ellensp, 2022-11-26:
PR created to fix this

closed 2022-11-26 by ellensp