MarlinFirmware/Marlin #25018: [BUG] No PWM on Pins 5, 6 and 44 (LED RGBW) ATmega2560 on MKS Gen L V2.0

aviatorhh, 2022-11-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Dear Development Team,

I have connected a LED RGBW strip via a transistor board to the Pins 5 (R), 4(G), 6(B) and 44 (W) and enabled RGBW_LED. When setting the values for LED lighting via M150 to S255 the corresponding LED illuminates. Setting a value of 254 or less the light switches off (0V at the pin). This does not affect pin 4 (G). This pin works as expected.
Even using M43 does not change the behaviour.
From past versions (v2.0.x) I can remember having pin 44 running on PWM before so I have seen PWM working on that board.

### Bug Timeline

Seems to be from version 2.1 on

### Expected behavior

I expected to dim all leds on pins 4, 5, 6 and 44 not on 4 alone.

### Actual behavior

Only pin 4 dims and the others do not react to PWM values lower 255

### Steps to Reproduce

e.g.
1. M150 R255 (red led is on)
2. M150 R254 (red led is off)

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Nov 25 2022 20:28:19)

### Printer model

Anet A8

### Electronics

MKS GEN L V2.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Using Slicer from Autodesk Fusion 360 (n/a to this issue)
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/10094959/Configs.zip)

ellensp, 2022-11-28:
Most of your pins are in conflict

R is on pin 5
as is SPINDLE_DIR_PIN

G is on pin 4
as is FAN_PIN, FIL_RUNOUT_PIN and SPINDLE_LASER_ENA_PIN

B is on pin 6 
as is SPINDLE_LASER_PWM_PIN and CASE_LIGHT_PIN

W is on 44
no conflicts

But even removing these pin conflicts, the PWM fails to function.

ellensp, 2022-11-28:
A git bisect says e3e07354d5ff98293f87d532fdd1c384d8e0b2e2 is the commit that broke it

ellensp, 2022-11-28:
reverting the change to Marlin/src/HAL/AVR/fast_pwm.cpp definitely fixes it

From 
```CPP
      /// Calculate frequencies of test prescaler and resolution values
      const int f_diff = ABS(f - int(f_desired)),
                f_fast_temp = (F_CPU) / (p * (1 + res_fast_temp)),
                f_fast_diff = ABS(f_fast_temp - int(f_desired)),
                f_pc_temp = (F_CPU) / (2 * p * res_pc_temp),
                f_pc_diff = ABS(f_pc_temp - int(f_desired));
```
back to 
```CPP
      // Calculate frequencies of test prescaler and resolution values
      const uint32_t f_diff      = _MAX(f, f_desired) - _MIN(f, f_desired),
                     f_fast_temp = (F_CPU) / (p * (1 + res_fast_temp)),
                     f_fast_diff = _MAX(f_fast_temp, f_desired) - _MIN(f_fast_temp, f_desired),
                     f_pc_temp   = (F_CPU) / (2 * p * res_pc_temp),
                     f_pc_diff   = _MAX(f_pc_temp, f_desired) - _MIN(f_pc_temp, f_desired);
```

just need to look into why exactly....

ellensp, 2022-11-28:
This fixes it.

```DIFF
diff --git a/Marlin/src/HAL/AVR/fast_pwm.cpp b/Marlin/src/HAL/AVR/fast_pwm.cpp
index e440315e00..3f17de5095 100644
--- a/Marlin/src/HAL/AVR/fast_pwm.cpp
+++ b/Marlin/src/HAL/AVR/fast_pwm.cpp
@@ -146,7 +146,7 @@ void MarlinHAL::set_pwm_frequency(const pin_t pin, const uint16_t f_desired) {
       LIMIT(res_pc_temp, 1U, maxtop);
 
       // Calculate frequencies of test prescaler and resolution values
-      const int f_diff = ABS(f - int(f_desired)),
+      const uint32_t f_diff = ABS(f - int(f_desired)),
                 f_fast_temp = (F_CPU) / (p * (1 + res_fast_temp)),
                 f_fast_diff = ABS(f_fast_temp - int(f_desired)),
                 f_pc_temp = (F_CPU) / (2 * p * res_pc_temp),
```

aviatorhh, 2022-11-28:
FYI my Pin debug. It may was confusing because I did not send in the changed pin definition files (MKS/RAMPS).

Recv: PIN:   4   Port: G5        RGB_LED_G_PIN                          Output = 0    TIMER0B   PWM:     1    WGM: 3    COM0B: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 3   overflow interrupt enabled

Recv: PIN:   5   Port: E3        RGB_LED_R_PIN                          Output = 0    TIMER3A   PWM:     1    WGM: 1    COM3A: 1    CS: 0    TCCR3A: 1    TCCR3B: 0    TIMSK3: 0

Recv: PIN:   6   Port: H3        RGB_LED_B_PIN                          Output = 0    TIMER4A   PWM:     1    WGM: 1    COM4A: 1    CS: 0    TCCR4A: 1    TCCR4B: 0    TIMSK4: 0

Recv: PIN:  44   Port: L5        RGB_LED_W_PIN                          Output = 1    TIMER5C   PWM:     1    WGM: 1    COM5C: 0    CS: 0    TCCR5A: 1    TCCR5B: 0    TIMSK5: 0

ellensp, 2022-11-28:
A PR to fix this issue has been created. Closing this issue.

aviatorhh, 2022-11-28:
I can confirm the fix did it (at least on my side). All works as expected now. I tested manually with M150 and can also see the startup test going thru "all" colors. Great work! Thanks!