MarlinFirmware/Marlin #25022: [FR] (feature summary)

BossDenya, 2022-11-26:
### Is your feature request related to a problem? Please describe.

Hello. I have an Ender 3s1 pro printer. Marlin does not support screen on this printer. Please make MARLIN work with Ender 3s1 pro touch screen

### Are you looking for hardware support?

Hello. I have an Ender 3s1 pro printer. Marlin does not support screen on this printer. Please make MARLIN work with Ender 3s1 pro touch screen

### Describe the feature you want

Hello. I have an Ender 3s1 pro printer. Marlin does not support screen on this printer. Please make MARLIN work with Ender 3s1 pro touch screen

Pethical, 2023-01-01:
You can ask Creality to push back to Marlin their modifications. Ah, just before, you may ask their to do not violate Marlin GPL license and publish the source code. :-)

Pethical, 2023-01-15:
And voala, here is the source code for S1 Pro: https://github.com/CrealityOfficial/Ender-3S1/pull/35