MarlinFirmware/Marlin #25028: [BUG] bug in folders within folders system

ViniFariaXD, 2022-11-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If we have a folder with gcodes inside, it works perfectly. However, if we have a folder with another folder inside, and this one with the gcodes, the system crashes.
However, this system worked very well in previous versions. But I can't define exactly which version, because it used a firmware provided by Two Trees for Bluer.

### Bug Timeline

new, I don't know.

### Expected behavior

How it should be:
https://user-images.githubusercontent.com/119058777/204153258-4a702b96-5864-43ac-a723-261085344deb.mp4

### Actual behavior

How it is:
https://user-images.githubusercontent.com/119058777/204153248-dc6b1758-122b-4e15-a9b0-e09764241bd2.mp4

### Steps to Reproduce

folders within folders and gcodes
print

### Version of Marlin Firmware

2.1.1

### Printer model

Two Trees Bluer

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10098941/Marlin.zip)

NNGB-kor, 2022-11-29:
My marlin also can't run the gcode in the folder.
It has been like that for a long time.
That's why I had to put all the gcode at the " F:\ "  of the SD card.

ViniFariaXD, 2022-11-29:
> Meu marlin também não pode executar o gcode na pasta. Tem sido assim há muito tempo. É por isso que eu tive que colocar todo o gcode no " F:\ " do cartão SD.

I had to stop using the LVGL UI mode, because in addition to the failure of the folders, my Y axis was also losing step.

closed 2023-02-08 by github-actions[bot]