MarlinFirmware/Marlin #25032: Improve documentation about multiple Z motors numbering and layout

dasaki, 2022-11-29:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Im building a prototype printer with 4 independent Z motors.

I defined:
#define AUTO_BED_LEVELING_BILINEAR
and
`#define Z_STEPPER_ALIGN_XY { {  100, 100 }, { 800,  100 }, { 800, 800 }, { 100, 800 } }
`
I arranged the motors counter-clockwise, as seems suggested by the default Z_STEPPERS_ORIENTATION ("0") schematic:

` * 4 Steppers:  (0) 
   *               | 4   3 | 
   *               |       | 
   *               | 1   2 | 
   `
As seen from above, I assume X and Y orientation is the following:
`
    +Y        | 4   3 | 
    ^          |       | 
    |          | 1   2 | 

              -----> +X
`

However this leaded to an "accuracy decreasing" warning when trying to align multiple Z steppers using a bed probe ( G34 command ).
So I change the arrangement of the motors to clockwise:
`   *               | 2   3 |
   *               |       |
   *               | 1   4 |`

Then the aligment worked without warnings with the same Z_STEPPER_ALIGN_XY positions:  { {  100, 100 }, { 800,  100 }, { 800, 800 }, { 100, 800 } }

I couldn't find any documentation about the default motor layout/numbering in the documentation besides the comment for Z_STEPPERS_ORIENTATION in the Configuration_adv.h file.  

Is there anything I'm doing wrong? 

It would be nice to have some documentation about the default 3 and 4 z-stepper layout/numberting.

Thanks.

### Bug Timeline

Old issue

### Expected behavior

I expected to do the auto z motor alignment propery with the 4 motors arranged counter-clockwise as per  the comment for Z_STEPPERS_ORIENTATION

### Actual behavior

It issues a "ACCURACY DECREASING" warning

### Version of Marlin Firmware

CONFIGURATION_H_VERSION 02010100

### Printer model

custom

### Electronics

BTT Octopus

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

 
[Marlin_211_configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/10111850/Marlin_211_configuration_files.zip)

InsanityAutomation, 2022-11-29:
You are overriding the values defined by #define Z_STEPPERS_ORIENTATION 0 with #define Z_STEPPER_ALIGN_STEPPER_XY being manually set. The values you set align with orientation 0, so I would question if each port is what you think it is electrically. 

Can you use G34Z to lock steppers and jog 1 at a time?

 * Manual stepper lock controls (reset by G28):
 *   L                 Unlock all steppers
 *   Z<1-4>            Z stepper to lock / unlock
 *   S<state>          0=UNLOCKED 1=LOCKED. If omitted, assume LOCKED.
 *
 *   Examples:
 *     G34 Z1     ; Lock Z1
 *     G34 L Z2   ; Unlock all, then lock Z2
 *     G34 Z2 S0  ; Unlock Z2
 *

dasaki, 2022-11-30:
>    - S 0=UNLOCKED 1=LOCKED. If omitted, assume LOCKED.

I tested the locking thing ( whitch I didn't know about before, even afte
googling "marlin move single z motor", etc. ), and my electrical wiring is
ok, so my current setup is clockwise:

2      3

1      4

However the Z_STEPPER_ALIGN_STEPPER_XY values I defined ( { { 100, 100 }, {
800, 100 }, { 800, 800 }, { 100, 800 } } ) are ordered conter-clockwise. I
defined them because I didn't find info about the how are the positions
calculated automatically with Z_STEPPERS_ORIENTATION, and was afraid of the
printer trying to reach some MAX_X and MAX_Y endstop, which I dont have
(sorry for my noobness).

Thanks for your help :)

dasaki, 2022-11-30:
Ok, went back to the counter-clockwise motor arrangement and it does show
the "accuracy decreasing" again.

dasaki, 2022-12-01:
Also, I've noticed the 900*900mm bed is slightly warped, the center is
about 1mm higher than the corners. Might this affect the alignment? or does
the Code compensate for bed warping on the fly while printng/moving?

closed 2023-02-10 by github-actions[bot]