MarlinFirmware/Marlin #25037: [FR] Modify "Hotend Idle Timeout" on gui or by commands

Zincr0, 2022-11-30:
### Is your feature request related to a problem? Please describe.

Currently, the "Hotend Idle Timeout" cannot by disabled or its value modified without recompiling.

It's unclear if this timeout is the same as https://marlinfw.org/docs/gcode/M085.html 

This timeout shutdowns not only the Hotend temperature but also the bed temperature, which is at least misleading
 and can easily ruin prints if it ends up unsticking the print from bed. 

There's no way (or at least it's not documented) to know what the actual value is by commands.  

On some printer configurations, it's recomended to warm the bed for at least 10 minutes. This seems to be just about the value from "Hotend Idle Timeout" ("seems" because there's no documented way to know what the actual value is) so it's simple useless for those printers. 

### Describe the feature you want

Document if the "Hotend Idle Timeout" is the same as https://marlinfw.org/docs/gcode/M085.html or add some way to modify it without recompiling. Also separate the bed temperature timeout from the Hotend temperature timeout.

thisiskeithb, 2022-11-30:
See https://github.com/MarlinFirmware/Marlin/pull/25015

thisiskeithb, 2022-11-30:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23329

closed 2022-11-30 by thisiskeithb

Zincr0, 2022-11-30:
"Duplicate of #23329" doesn't answer most of this request, 
It is the same as the M085? 
can it be separated the bed temperature from the hotend temperature? 
Is there a way to check what the actual value is (view the value, not modifyng it)?
Did you even read this feature before closing it?