MarlinFirmware/Marlin #25044: [FR] G-Code Debugging Symbols for a JIT Debugger & dynamic reslicing

dcunited001, 2022-12-01:
### Is your feature request related to a problem? Please describe.

Slicing should be editable so large prints on small printers don't have to be recycled when one part makes the whole print fail.

### Describe the feature you want

You should be able to pause the print, connect to a web app/debugger, see the gcode "AST" objects (which contain debugging information that identifies them as symbols, not commands) and eliminate the remainder of gcode lines associated with particular objects.

The gcode mid-execution should be broken in two sections (before/after), and interpolation instructions provided allowing the print to resume -- this is essentially the "pause" functionality in the Ender 3 Pro.  Then the subsequent sections are updated removing the problematic objects or allowing the injection of a header to execute before the print resumes (e.g. to fix a problematic object, which to some extent one can already do with a 3D doodle pen). 

Either: 

- the marlin g-code flavor should be adapted to inject identifying information relating to separate parts (this is a tensor deconstruction on the sliced path, rolling it back into distinctly INDEXable objects)
- or the firmware should allow the memory image to be dynamically updated, where the slicer (in conjunction with the post-processor i guess) simply replace the rest of the gcode in memory.

I guess it depends on whether the device is embedded or not.

ellensp, 2022-12-01:
Marlin is for micro controllers, a lot of which are still 8 bit with 4k of ram 
Most don't have web interfaces, no operating systems, no way to do anything like this.

dcunited001, 2023-02-09:
i didn't know about that. thanks @thisiskeithb

closed 2023-06-06 by dcunited001