MarlinFirmware/Marlin #25046: [FR] Support TFT SPI screens on STM32H7 pratform

vovodroid, 2022-12-01:
### Is your feature request related to a problem? Please describe.

Currently "[SPI TFT is currently only supported on STM32F1 and STM32F4 hardware](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/HAL/STM32/tft/tft_spi.h#L29)", while new BTT SKR 3 board is STM32H7 based. Thus it's impossible to replace stock BTT SKR 2 board with SKR 3, as Biqu S1 SE uses TFT35 SPI screen.

I tried to add `#include "stm32h7xx_hal.h"`, but build failed with error 

```
Marlin\src\HAL\STM32\tft\xpt2046.cpp:146:18: error: 'struct SPI_TypeDef' has no member named 'DR'; did you mean 'SR'?
  146 |   SPIx.Instance->DR = data;
```
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/HAL/STM32/tft/xpt2046.cpp#L146

### Are you looking for hardware support?

Biqu S1 SE (Plus)

### Describe the feature you want

Support TFT SPI screens on STM32H7 pratform

jmz52, 2022-12-02:
@vovodroid, will you be able to test various development versions of code?
I can work on this, but I do not have STM32H7 board, so I need someone to test my code.

vovodroid, 2022-12-02:
> will you be able to test

No, I don't have STM32H7 boards, but it would good to have compiled code ready.

ellensp, 2022-12-04:
@jmz52 I have a BTT SKR 3 and a Makerbase MKS TS35 3.5 (SPI interface) and would be able to do some testing.

vovodroid, 2022-12-04:
@ellensp Which screen do you use now with SKR 3?

ellensp, 2022-12-04:
@vovodroid none, I only got it a few days ago, not in use yet. Thus can be used for testing for a while.

thisiskeithb, 2022-12-04:
> I have a BTT SKR 3 and a Makerbase MKS TS35 3.5 V2 (SPI interface)

Ditto.

I also have a Biqu SE Plus with the BTT TFT35 SPI TFT (`BTT_TFT35_SPI_V1_0`) that I can swap an SKR 3/EZ into for testing.

On a related note, the BTT SPI screen is not for sale as a standalone product currently, so it’d be a really limited upgrade path/combo when used on the SE Plus with an SKR 3.

Since MKS sells their SPI TFT as a standalone product at retail, this FR would still make sense to get working On H7 mcus.

quiret, 2022-12-04:
https://github.com/MarlinFirmware/Marlin/pull/24911

I want to work on this, too. Not sure how generic my implementation is but I am willing to read the docs and get a really good one done.

jmz52, 2023-01-04:
[image]
Halfway there.
It took quite some time (and a new dev board) to make some sense of H7 architecture, but here it is.
Please mind that this is a teaser - code is not fully completed and I still need to merge new H7 code with the original F1/F4 code.

jmz52, 2023-02-04:
Code is ready for testing - https://github.com/jmz52/Marlin/tree/H7-wip
Configuration files are from "BIQU/B1 SE", with mainboard changed to BTT_SKR_V3_0 and disabled USB flash drive support. 

TFT SPI code works fine, but touch screen support was not tested beyond mere compilation.
I need to do some soldering to connect MKS TS35 to my dev board and I don't have a slightest idea where did I put those 2.54mm pins headers.

thisiskeithb, 2023-02-11:
> Code is ready for testing - https://github.com/jmz52/Marlin/tree/H7-wip
> Configuration files are from "BIQU/B1 SE", with mainboard changed to BTT_SKR_V3_0 and disabled USB flash drive support.

I changed the motherboard to `BOARD_BTT_SKR_V3_0_EZ` since that's what I have hooked up right now, but I only get a solid white screen with a BTT TFT35 SPI V1.0 (`BTT_TFT35_SPI_V1_0` that's defined in the config).

jmz52, 2023-02-11:
> I hooked up an MKS TS35 V2.0 (`MKS_TS35_V2_0`) and I also get a solid white screen.

Just in case - have you updated TFT and touch pins definitions for MKS TS35?

thisiskeithb, 2023-02-11:
> Just in case - have you updated TFT and touch pins definitions for MKS TS35?

I commented out `HAS_SPI_TFT` & used this pinout from the Nano V3 pins file:

```cpp
#if ENABLED(MKS_TS35_V2_0)

  #define TFT_CS_PIN                 EXP1_07_PIN
  #define TFT_SCK_PIN                EXP2_02_PIN
  #define TFT_MISO_PIN               EXP2_01_PIN
  #define TFT_MOSI_PIN               EXP2_06_PIN
  #define TFT_DC_PIN                 EXP1_08_PIN
  #define TFT_A0_PIN                  TFT_DC_PIN

  #define TFT_RESET_PIN              EXP1_04_PIN

  #define LCD_BACKLIGHT_PIN          EXP1_03_PIN
  #define TFT_BACKLIGHT_PIN    LCD_BACKLIGHT_PIN

  #define TOUCH_BUTTONS_HW_SPI
  #define TOUCH_BUTTONS_HW_SPI_DEVICE          1

  #ifndef TFT_WIDTH
    #define TFT_WIDTH                        480
  #endif
  #ifndef TFT_HEIGHT
    #define TFT_HEIGHT                       320
  #endif

  #define TOUCH_CS_PIN               EXP1_05_PIN  // SPI1_NSS
  #define TOUCH_SCK_PIN              EXP2_02_PIN  // SPI1_SCK
  #define TOUCH_MISO_PIN             EXP2_01_PIN  // SPI1_MISO
  #define TOUCH_MOSI_PIN             EXP2_06_PIN  // SPI1_MOSI

  #define LCD_READ_ID                       0xD3
  #define LCD_USE_DMA_SPI

  #define TFT_BUFFER_SIZE                  14400

  #ifndef TOUCH_CALIBRATION_X
    #define TOUCH_CALIBRATION_X           -17253
  #endif
  #ifndef TOUCH_CALIBRATION_Y
    #define TOUCH_CALIBRATION_Y            11579
  #endif
  #ifndef TOUCH_OFFSET_X
    #define TOUCH_OFFSET_X                   514
  #endif
  #ifndef TOUCH_OFFSET_Y
    #define TOUCH_OFFSET_Y                   -24
  #endif
  #ifndef TOUCH_ORIENTATION
    #define TOUCH_ORIENTATION    TOUCH_LANDSCAPE
  #endif

#endif
```

> Also MKS TS35 needed reversed cables

Yup. I'm quite familiar with this issue since it causes a lot of headaches in Marlin's EXP1/2 numbering 🙂 Most of my EXP cables have the tabs shaved off because of it.

jmz52, 2023-02-11:
@thisiskeithb, please check following configuration:
In `Configuration.h` add `#define TFT_DRIVER ST7796`
Change `SPIx.Init.BaudRatePrescaler` to `SPI_BAUDRATEPRESCALER_64` in https://github.com/jmz52/Marlin/blob/H7-wip/Marlin/src/HAL/STM32/tft/tft_spi.cpp#L64

thisiskeithb, 2023-02-11:
> In `Configuration.h` add `#define TFT_DRIVER ST7796`

Shouldn't that be set via conditionals when `BTT_TFT35_SPI_V1_0` is enabled? :

https://github.com/MarlinFirmware/Marlin/blob/ef3f484d3ca620272f849b913851541b54b51c09/Marlin/src/inc/Conditionals_LCD.h#L1474-L1478

I defined it anyway and updated `SPI_BAUDRATEPRESCALER_4` to `SPI_BAUDRATEPRESCALER_64`, but I still get a white screen with `BTT_TFT35_SPI_V1_0`.

jmz52, 2023-02-12:
The are a bit different.
`TFT_DEFAULT_DRIVER` is a fall-back value is case driver detection fails.
`TFT_DRIVER` force the result, bypassing detection completely.

As I've found the H7 driver detection code only worked in half-duplex SPI (used by ST7735) but it hangs in full-duplex mode.

With updated code my TS35 is now displays proper UI, but touch is not working. This could be either code issue or wiring mess issue.
@thisiskeithb, please check if there are any improvements with your screens.

thisiskeithb, 2023-02-22:
> please check if there are any improvements with your screens

Success!

We'll have to tweak some pins files to differentiate between `MKS_TS35_V2_0` & `BTT_TFT35_SPI_V1_0` since `HAS_SPI_TFT` does not work for both ([per my comment above](https://github.com/MarlinFirmware/Marlin/issues/25046#issuecomment-1426845885)).

Touch also worked on the `BTT_TFT35_SPI_V1_0` once I remembered to reset the EEPROM to clear out the saved values from the MKS build 😆

thisiskeithb, 2023-04-07:
@jmz52: Other than refactoring / fixing the `HAS_SPI_TFT` macro in the various pins files to separate MKS & BTT SPI TFTs, is there anything you're waiting on for a PR? I plan to replace the SKR 2 in my B1 SE Plus with an SKR 3, so your work on the SPI TFT code will be a welcome addition!

jmz52, 2023-04-07:
I was waiting for https://github.com/MarlinFirmware/Marlin/pull/25359 to avoid conflicts in initialization section
Now it is merged and next week I should have time to create PR for H7

thisiskeithb, 2023-05-03:
> Please retest code from new branch - https://github.com/jmz52/Marlin/tree/stm32h7-tft-spi

Paired with my to-be-submitted PR to split up the `HAS_SPI_TFT` pins, this *mostly* works with `BTT_TFT35_SPI_V1_0` & `MKS_TS35_V2_0`.

I say mostly, because `SPEAKER` on `MKS_TS35_V2_0` causes the graphics to go haywire when you send a tone with `M300`. Clicking the encoder or touching the TFT fixes it, but I'm trying some different pin options to see if I can get that to stop.

 `SPEAKER` on `BTT_TFT35_SPI_V1_0` works fine.

Related to this, we still need https://github.com/MarlinFirmware/Marlin/pull/25579 to fix the issue with calibration / oritentation values from being ignored in pins files.

Here's the `pins_BTT_SKR_V3_0_common.h` diff required to support both `BTT_TFT35_SPI_V1_0` & `MKS_TS35_V2_0`:
<details><summary>expand me:</summary>
<p>

```diff
diff --git a/Marlin/src/pins/stm32h7/pins_BTT_SKR_V3_0_common.h b/Marlin/src/pins/stm32h7/pins_BTT_SKR_V3_0_common.h
index 09b305f..38f3b1e 100644
--- a/Marlin/src/pins/stm32h7/pins_BTT_SKR_V3_0_common.h
+++ b/Marlin/src/pins/stm32h7/pins_BTT_SKR_V3_0_common.h
@@ -455,6 +455,92 @@
     #define BTN_EN1                  EXP2_03_PIN
     #define BTN_EN2                  EXP2_05_PIN
 
+  #elif HAS_SPI_TFT                               // Config for Classic UI (emulated DOGM) and Color UI
+
+    #define TFT_SCK_PIN              EXP2_02_PIN
+    #define TFT_MISO_PIN             EXP2_01_PIN
+    #define TFT_MOSI_PIN             EXP2_06_PIN
+
+    #define BTN_EN1                  EXP2_03_PIN
+    #define BTN_EN2                  EXP2_05_PIN
+
+    #ifndef TFT_WIDTH
+      #define TFT_WIDTH                    480
+    #endif
+    #ifndef TFT_HEIGHT
+      #define TFT_HEIGHT                   320
+    #endif
+
+    #if ENABLED(BTT_TFT35_SPI_V1_0)
+
+      #define TFT_CS_PIN             EXP2_04_PIN
+      #define TFT_DC_PIN             EXP2_07_PIN
+      #define TFT_A0_PIN              TFT_DC_PIN
+
+      #define TOUCH_INT_PIN          EXP1_07_PIN
+      #define TOUCH_MISO_PIN         EXP1_06_PIN
+      #define TOUCH_MOSI_PIN         EXP1_03_PIN
+      #define TOUCH_SCK_PIN          EXP1_05_PIN
+      #define TOUCH_CS_PIN           EXP1_04_PIN
+
+      #ifndef TOUCH_CALIBRATION_X
+        #define TOUCH_CALIBRATION_X        17540
+      #endif
+      #ifndef TOUCH_CALIBRATION_Y
+        #define TOUCH_CALIBRATION_Y       -11388
+      #endif
+      #ifndef TOUCH_OFFSET_X
+        #define TOUCH_OFFSET_X               -21
+      #endif
+      #ifndef TOUCH_OFFSET_Y
+        #define TOUCH_OFFSET_Y               337
+      #endif
+      #ifndef TOUCH_ORIENTATION
+        #define TOUCH_ORIENTATION TOUCH_LANDSCAPE
+      #endif
+
+    #elif ENABLED(MKS_TS35_V2_0)
+
+      #define TFT_CS_PIN             EXP1_07_PIN
[... 159 lines cut ...]
 //
 // Touch Screen Settings
 //
-//#define TOUCH_SCREEN
+#define TOUCH_SCREEN
 #if ENABLED(TOUCH_SCREEN)
   #define BUTTON_DELAY_EDIT      50 // (ms) Button repeat delay for edit screens
   #define BUTTON_DELAY_MENU     250 // (ms) Button repeat delay for menus
diff --git a/platformio.ini b/platformio.ini
index e3bdb6f..12f2fc4 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -13,7 +13,7 @@
 [platformio]
 src_dir      = Marlin
 boards_dir   = buildroot/share/PlatformIO/boards
-default_envs = mega2560
+default_envs = STM32H723Vx_btt
 include_dir  = Marlin
 extra_configs =
     Marlin/config.ini
```

</p>
</details>

thisiskeithb, 2023-05-05:
Closing since there's now a PR to add support: 

- https://github.com/MarlinFirmware/Marlin/pull/25784

closed 2023-05-05 by thisiskeithb