MarlinFirmware/Marlin #25047: Fail to compile marlin on btt skr mini e3 v3

ardiramelan, 2022-12-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi guys, im starting to pull my hair now. Whenever i try to build the marlin firmware for skr mini e3 v3 using vs code, i always get this errror

FileNotFoundError: [Errno 2] No such file or directory: 'C:\Users\rahardi.kramadibrata\.platformio\.cache\tmp\pkg-installing-5fvbqsuq\Arduino_Core_STM32-main\variants\STM32MP1xx\MP133AA(E-F-G)_MP133CA(E-F-G)_MP133DA(E-F-G)_MP133FA(E-F-G)_MP135AA(E-F-G)_MP135CA(E-F-G)_MP135DA(E-F-G)_MP135FA(E-F-G)\variant_generic.cpp'.

however, when I tried several times using skr mini e3 v2 or any other board such as ANET board it worked. I wonder what is the problem ? any help would be appreciated

### Bug Timeline

1 December 2022

### Expected behavior

 Success build/compile the firmware

### Actual behavior

 Can't build/compile the firmware

### Steps to Reproduce

1. Download and extract Marlin 2.1.1
2. Edit the header file (configuration .h) #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V3_0 
3. Edit the platformio.ini default_envs = STM32G0B1RE_btt
4. Build / Compile the firmware

### Version of Marlin Firmware

2.1.1

### Printer model

Creality Ender 3

### Electronics

Bigtreetech SKR Mini E3 V3 

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

`` *  Executing task in folder Marlin-2.1.1: C:\Users\rahardi.kramadibrata\.platformio\penv\Scripts\platformio.exe run 

Processing STM32G0B1RE_btt (platform: ststm32@~14.1.0; board: marlin_STM32G0B1RE; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip
Unpacking  [##################################--]   97%  00:00:00
Unpacking...
Error: Traceback (most recent call last):
  File "C:\Users\rahardi.kramadibrata\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 38, in unpack
    return fu.unpack(dst, with_progress=with_progress)
  File "C:\Users\rahardi.kramadibrata\.platformio\penv\lib\site-packages\platformio\package\unpack.py", line 175, in unpack     
    self._archiver.extract_item(item, dest_dir)
  File "C:\Users\rahardi.kramadibrata\.platformio\penv\lib\site-packages\platformio\package\unpack.py", line 48, in extract_item    self._afo.extract(item, dest_dir)
  File "C:\Users\rahardi.kramadibrata\.platformio\python3\lib\zipfile.py", line 1616, in extract
    return self._extract_member(member, path, pwd)
  File "C:\Users\rahardi.kramadibrata\.platformio\python3\lib\zipfile.py", line 1687, in _extract_member
    open(targetpath, "wb") as target:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\rahardi.kramadibrata\\.platformio\\.cache\\tmp\\pkg-installing-p0hqc3oj\\Arduino_Core_STM32-main\\variants\\STM32MP1xx\\MP133AA(E-F-G)_MP133CA(E-F-G)_MP133DA(E-F-G)_MP133FA(E-F-G)_MP135AA(E-F-G)_MP135CA(E-F-G)_MP135DA(E-F-G)_MP135FA(E-F-G)\\variant_generic.cpp'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... 131 lines cut ...]
    self._archiver.extract_item(item, dest_dir)
  File "C:\Users\rahardi.kramadibrata\.platformio\penv\lib\site-packages\platformio\package\unpack.py", line 48, in extract_item    self._afo.extract(item, dest_dir)
  File "C:\Users\rahardi.kramadibrata\.platformio\python3\lib\zipfile.py", line 1616, in extract
    return self._extract_member(member, path, pwd)
  File "C:\Users\rahardi.kramadibrata\.platformio\python3\lib\zipfile.py", line 1687, in _extract_member
    open(targetpath, "wb") as target:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\rahardi.kramadibrata\\.platformio\\.cache\\tmp\\pkg-installing-i94wjf7m\\Arduino_Core_STM32-main\\variants\\STM32MP1xx\\MP133AA(E-F-G)_MP133CA(E-F-G)_MP133DA(E-F-G)_MP133FA(E-F-G)_MP135AA(E-F-G)_MP135CA(E-F-G)_MP135DA(E-F-G)_MP135FA(E-F-G)\\variant_generic.cpp'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

 *  The terminal process "C:\Users\rahardi.kramadibrata\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

jmz52, 2022-12-01:
Are there non-english characters in your windows username?

thisiskeithb, 2022-12-01:
Delete your `C:\Users\rahardi.kramadibrata\.platformio\packages` & `C:\Users\rahardi.kramadibrata\.platformio\.cache` folders and let PIO rebuild them.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2022-12-01 by thisiskeithb

ardiramelan, 2022-12-02:
For anybody who share the same issue, it is because the folder name is too long. The solution is to enable the longpath in the windows32 

see the tutorial here :https://www.thewindowsclub.com/how-to-enable-or-disable-win32-long-paths-in-windows-11-10