MarlinFirmware/Marlin #25049: PROBABLE MARLIN FIMWARE GPL V3 VIOLATION

giutrec, 2022-12-02:
### Describe the feature you want

I would like to inform you that this youtuber creator declared that he was using a modified version of MarlinFirmware to be able to 3D print with a MIG/MAG welding machine when I proceeded to inform him in a comment that the "secret" code he uses is a code released by the GNU GPL v3 and that should release the code and make it available in accordance with the license itself, proceeded to remove my comment.

The video where they admit that they use Marlin firmware

[[www.youtube.com](https://www.youtube.com/watch?v=5y1bBoih_sE)]

thisiskeithb, 2022-12-02:
Maybe I missed something in the video since I was using auto-translated Italian->English captions, but they didn't hide the fact that it was running marlin. There is even some b-roll with Marlin booting/running.

This appears to be a work-in-progress / on-off project for a thesis and not something generally/commercially available. Are there plans to sell or commercialize a machine running Marlin?

Have you asked them for a copy of the source code?

EvilGremlin, 2022-12-02:
@giutrec Anyone is free to do whatever they want to GPL code (or any other license) as long as they don't distribute binaries in any form. Only then providing source is required. He did not violate anything, issue closed.

GMagician, 2022-12-02:
Yes it's a thesis project and author clearly stated that he used Marlin. It seems that there is something related to the head that may become something commercial, but on final of the video it seems that may be something more also related to the whole machine.

closed 2022-12-02 by thisiskeithb