MarlinFirmware/Marlin #25051: [FR] Add little extrusion prior to retraction when changing filament

adrianmihalko, 2022-12-03:
### Is your feature request related to a problem? Please describe.

I am going to ask to reopen FR #16545. It would be super nice to implement this feature, because current unload method is not working together for example with BMG extruder.

@yet-another-average-joe nicely summarized the issue in his comment:
 https://github.com/MarlinFirmware/Marlin/issues/16545#issuecomment-573420029

...also provided a solution.

### Describe the feature you want

Extrude filament before unload filament.

EvilGremlin, 2022-12-03:
It's all there already for a long time
``` 
  #define FILAMENT_UNLOAD_PURGE_RETRACT       13  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_PURGE_DELAY       5000  // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8  // (mm) An unretract is done, then this length is purged.
  #define FILAMENT_UNLOAD_PURGE_FEEDRATE      25  // (mm/s) feedrate to purge before unload
```

adrianmihalko, 2022-12-03:
> It's all there already for a long time
>
> ```
>   #define FILAMENT_UNLOAD_PURGE_RETRACT       13  // (mm) Unload initial retract length.
>   #define FILAMENT_UNLOAD_PURGE_DELAY       5000  // (ms) Delay [...]

Oh super. I don't understand the retract before purge, currently I am using unload script:

```
G21 ;Let the printer know you are following this point with metric values
G90 ;Let the printer know you are using absolute positioning
M109 S200.000 ;Heat hotend to 200C for PLA ‘S245.000’ for ABS
G92 E0 ;Reset the position of the extruder
G1 E20 F100 ; Extrude a short distance before unloading to avoid blob forming
G92 E0 ;Reset the position of the extruder
G1 E-700 F2000 ;Retract 700 mm of filament at 2000 mm/minute speed, change 700 to the lenght of your bowdentube + 100 mm.
G92 E0 ;Reset the position of the extruder
M117 REMOVE FILAMENT NOW ;Display message on LCD-display to remove the filament
```

Should I leave PURGE_RETRACT as it is and/or set to 0 and increase PURGE_LENGTH to 20?

closed 2022-12-03 by thisiskeithb