MarlinFirmware/Marlin #25052: [FR] (feature summary)

lunartester, 2022-12-04:
### Is your feature request related to a problem? Please describe.

Homing X,Y and resetting Absolute Coordinates on X,Y,Z

### Are you looking for hardware support?

Creality 32-bit board or any other 32-bit board that supports Marlin and has 5V power for 3-wire optical end stop set up.

### Describe the feature you want

Markforged does not use encoders or any error correction nor any closed loop systems in their desktop series, they use optical end stops and their printers home to X,Y every few printing layers. Stratasys does the same thing even with encoders and closed loop systems. I suspect they do a some type of reset on error accumulation with that home X,Y. 

It's unclear if they use Absolute or Relative on X,Y,Z....... but that feature is the one that produces consistent results in compensation for any backlash or worn out gantry parts over long haul. Markforged layer seam is fixed position, the end user cannot change it in their slicer, so they know the exact distance from start of each layer to the X,Y,Z  end stops, or at least they can reset it every time it homes.  

Homing will require a separate wipe as last move before it homes and second retraction tied up to that specific procedure, so even if wipe is not selected in slicer there must be a wipe option for the end user and ability to set wipe distance and ability to set differnt retraction for that homing reset.

What is unclear to me is what is the exact algorithm they use there on error correction, is it just a reset, or something more complex. They use 64-bit beagelbone running on Linux, I want to use Creality 32-bit board.

I was told by Prusa dev team member implementing it in Slicer via gcode or script is not optimal and needs to be done in firmware. 

So the main question is, how trivial is it to add this to Marlin. I know it Marlin does a reset on Extruder with each layer change, can we expand this to X,Y,Z with homing and retraction/ wipe option for that homing or at elast the wipe and retraction setting can be changed via Gcode? 

The end user must have an option how often or after how many layers they want to reset it with homing.

Cheers.

ellensp, 2022-12-04:
Cannot be in firmware. Cannot tell that if the Z height is printing layer N or just jumped to layer N with something like a Z,hop.

Then there is non planer slicing, Where the moves are in x,y and z a lot. (not just Z on layers)   

In-fact I think whom ever suggested this should be firmware is pulling your leg. Or has no idea what they are talking about.

thisiskeithb, 2022-12-04:
> I know it Marlin does a reset on Extruder with each layer change

This is done in the slicer as well. Usually with a `G92 E0.0`:

[image]

lunartester, 2022-12-04:
resetting extruder without resetting X,Y is useless they must be time aligned. I wasn't asking about extruder I was asking about X,Y not sure on Z if they do it or not, sicne they do not home Z for obvious reasons. 

all commentary about fixing the problem of lost steps by other means then error correction or reset will be ignored, this is above your pay grade if you think fixing backlash compensation can be done by anything else then proper error correction. I mentioned two most reliable and most successful FDM pritners for a reason, don't care about any input or alternative solutions besides exact replica of what they are doing and how to do it in Marlin + Prusa Slicer. Also don't care about extruder reset, it's already implemented in Marlin. X,Y and Z is the key here, specially X,Y since it has belts.

Cheers

ellensp, 2022-12-04:
Your even talking cross talk

Marlin already has backlash compensation

It has nothing to do with re homing  and resetting position 0 at all

Backhand insults however will not be tolerated.

closed 2022-12-04 by ellensp

lunartester, 2022-12-04:
Can you please leave this discussion, you are wasting band width I want to hear from people that understand what I'm asking about and can simply answer my questions directly without offering any alternatives, this feature is not present in Marlin, alternative maybe, I don't care about alternatives, want to know how easy is it to do and how much it will cost as far as man hours.

thisiskeithb, 2022-12-04:
> I don't care about alternatives

I was correcting your assumption, not offering an alternative.

You said “I know it Marlin does a reset on Extruder with each layer change”

Marlin does not do this as it has no concept of layers. Extruder resets are done by your slicer. In the case of PrusaSlicer, that is done with the “Before layer change G-code” section. Cura and other slicers likely do something similar to prevent cumulative extruder distance overflows.

As ellensp stated, Marlin already supports backlash compensation (enabled with `BACKLASH_COMPENSATION`) which can be used to compensate for sloppy / worn out hardware.

[Read more about M425 - Backlash Compensation on marlinfw.org](https://marlinfw.org/docs/gcode/M425.html)