MarlinFirmware/Marlin #25054: [FR] (feature summary)

lunartester, 2022-12-04:
### Is your feature request related to a problem? Please describe.

Resetting X,Y on Homing

### Are you looking for hardware support?

32-bit Marlin board

### Describe the feature you want

Markforged does not use encoders or any error correction nor any closed loop systems in their desktop series, they use optical end stops and their printers home to X,Y every few printing layers. Stratasys does the same thing even with encoders and closed loop systems. I suspect they do a some type of reset on error accumulation with that home X,Y.

It's unclear if they use Absolute or Relative on X,Y,Z....... but that feature is the one that produces consistent results in compensation for any backlash or worn out gantry parts over long haul. Markforged layer seam is fixed position, the end user cannot change it in their slicer, so they know the exact distance from start of each layer to the X,Y,Z end stops, or at least they can reset it every time it homes.

Homing will require a separate wipe as last move before it homes and second retraction tied up to that specific procedure, so even if wipe is not selected in slicer there must be a wipe option for the end user and ability to set wipe distance and ability to set differnt retraction for that homing reset.

What is unclear to me is what is the exact algorithm they use there on error correction, is it just a reset, or something more complex. They use 64-bit beagelbone running on Linux, I want to use Creality 32-bit board.

I was told by Prusa dev team member implementing it in Slicer via gcode or script is not optimal and needs to be done in firmware and maybe something added in Gcode.

So the main question is, how trivial is it to add this to Marlin. I know it Marlin it does a reset on Extruder with each layer change, can we expand this to X,Y,Z with homing and retraction/ wipe option for that homing or at elast the wipe and retraction setting can be changed via Gcode?

The end user must have an option how often or after how many layers they want to reset it with homing.

Cheers.

ellensp, 2022-12-04:
Please update the Subject from "[FR] (feature summary) " to something related to the topic at hand

thisiskeithb, 2022-12-04:
> I know it Marlin does a reset on Extruder with each layer change

As I said in https://github.com/MarlinFirmware/Marlin/issues/25052#issuecomment-1336310110:

This is done in the slicer with a `G92 E0.0`:

[image]

Marlin does not do this as it has no concept of your print's layers. Extruder resets are done by your slicer. In the case of PrusaSlicer, that is done with the “Before layer change G-code” section. Cura and other slicers likely do something similar to prevent cumulative extruder distance overflows.

Also, as ellensp stated in https://github.com/MarlinFirmware/Marlin/issues/25052#issuecomment-1336312609, Marlin already supports backlash compensation (enabled with `BACKLASH_COMPENSATION`) which can be used to compensate for sloppy / worn out hardware.

[Read more about M425 - Backlash Compensation on marlinfw.org](https://marlinfw.org/docs/gcode/M425.html)

lunartester, 2022-12-04:
I do not need backlash compensation algorithm it doe snot work without closed loop feeback system of precise measures. I need that exact reset feature with homing procedure on X,Y and Z maybe works teh same way as Extruder reset on each layer, 

We already established that on X,Y and Z its not present in Marlin. I want to know what it would take to implement it on X,Y and maybe Z, (markforged does not use any Z-hope nor does Stratasys) .......how many man hours of programming.

Cheers.

EvilGremlin, 2022-12-04:
Then just use `G28 XY` in after layer change script in slicer

thisiskeithb, 2022-12-04:
> I need that exact reset feature with homing procedure on X,Y

If you want to reset your X and Y positions with a homing procedure, use the aforementioned “Before layer change G-code” feature of PrusaSlicer (and probably similarly through other slicers) to add a `G28 XY`.

> and Z

This is obviously a bad idea for Z unless you can home to Z max / away from a print.

> maybe works teh same way as Extruder reset on each layer

Since the extruder never “homes”, a reset can easily be issued via a G-code command.

lunartester, 2022-12-04:
How do we know that G28 does a reset vs just another move to home position and the errors are still accumulated after that? Then I also mentioned we need different retraction on homing and wipe before that as well and ability to decide after how man layers it does a home reset. The extruder reset has to align with X,Y reset otherwise it will start to extrude either earlier or later. 

So ideally I would not want Marlin's stock reset on each layer, maybe only when it homes, think about it, Extruder resets each layer yet each layer is not reset on X,Y only after few layers, so maybe it's better not to reset Extruder on each layer. 

I'm convinced markforged has the extruder reset aligned with X,Y reset somehow, they produce very clean prints with no blobs with homing and no nozzle brush on homing and it's bowden, that type of precision over 20,000 printing hours cannot be achieved with some type of time alignment on resetting extruder with X,Y to match them both.

lunartester, 2022-12-04:
Dumb question, can we combine relative with absolute, so for example it's printing in Absolute but that move to home is relative so naturally relative does a reset, then once it reaches the next layer starting position it goes back to absolute till next homing reset. This onyl solves one issue, the custom retraction on reset and wipe needs to be added as well.

EvilGremlin, 2022-12-04:
:facepalm: Because G28 is 'find home' command, not 'move to' command. You don't understand at all what you're talking about. Everything you want is easily done in slicer. You can script rehome every Nth layer, or every Nth millimeter, whatever.

lunartester, 2022-12-04:
What does mean find home vs reset to zero from home to seam position on next layer. The way I see it is marforged does not allow you to change seam angle or position, there is no random seam. So once you slice it the fimrware knows the exact distance from home to seam on each layer, so when ever it homes it resets it to 0,0 on X,Y and now it knows the exact distance to go to seam position. S

Stratasys allows you to change the seam position but they also use encoders so it's a different set up. Does Marlin or Prusa Slicer map out the whole print and knows the exact distance from home to layer seam on each layer?  So even if G28 does a reset, how do we introduce second retraction on that home and wipe before it homes? Can we reset the extruder via G28 as well

I see it this way, wipe first (custom wipe) then use second retraction much longer then stock one, then home, X,Y and reset X,Y,E  then move to new layer seam.

EvilGremlin, 2022-12-04:
What markforged and startasys do in their (stolen) code is of no consequence to Malrin.
There is no concept of "seam position" in any printer firmware. Seam only exist in slicer, after that it's all just simple moves. You confusing slicer peculiarities for firmware features.
> So even if G28 does a reset, how do we introduce second retraction on that home and wipe before it homes? Can we reset the extruder via G28 as well
Please stop spouting nonsense. Wipe and retract always happen before layer change. And there's absolutely no need to reset extruder, it's already being done by slicer if you use absolte E coordinates (G92). And if you use relative - there's obviously no need to reset anything.

lunartester, 2022-12-04:
Go over my full request, then open Prusa slicer under printer settings then under extruder, you are talking non-sense, not me. Firmware always overrides any slicer not the other way around, its done for safety reasons, so just because we can do something in slicer, does not mean Marlin will allow it or will override it. Homing and reset requires a different wipe and different retraction to hold it in for longer period of time, vs printing. a 6 year old can understand that my request on homing procedure needs it's own settings on speed, wipe retraction adn reset.

EvilGremlin, 2022-12-04:
Yeah, i glazed over it becase wipe is impossible to override in firmware. Only overrrides are absolute limits, and general FIRMWARE_RETRACT. Yeah, you can't really control last wipe before layer change as of now. Still this is 100% slicer responsibility to do. Firmware can't possibly know which move is a wipe before layer change, there is no lookahead or any advanced gcode parsing, firmware is dumb executor. And most importantly, firmware can't ever change any XYZ moves. Retract override is sole exception, for obvious reasons.

thisiskeithb, 2022-12-04:
> Firmware always overrides any slicer not the other way around

A slicer / G-code can absolutely override firmware settings. Again, it sounds like you have a misunderstanding of what each can do, so this request is not something that can be implemented / should be done via slicer G-code commands & scripting.

In addition to the links and info already posted, I’d recommend the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2022-12-04 by thisiskeithb

lunartester, 2022-12-04:
For some reason the people that are closing this topic have a major issue with large ego, small penis syndrome and lots of insecurity in between. If you do not know the answers then simply do not participate in the discussion , this issue has not been resolved, why are you closing this topic. I don't need reading material I
I'm not skilled to use that to produce a solution, I used the request option for a reason trying to find someone skilled enough to produce a script + Gcode in Prusa Slicer that I can use. or submit a request for Marlin to develop enough tools to accommodate that request.

The-EG, 2022-12-04:
@lunartester Why would anyone else want to help you given how you've responded to those that did already try? Good luck figuring it out.

lunartester, 2022-12-04:
Because human nature is based on 1%, .....99% are idiots only 1% will understand and will help, I'm gambling on stumbling on one person to help, 99% are idiots that get offended easily, you are adn the rest that are closing my topic are part of 99%, so I'll keep posting till I find one person from the 1%. 

Cheers.

EvilGremlin, 2022-12-04:
This behaviour will get you banned.

lunartester, 2022-12-04:
I don't care, I have a goal to get from A to B, if I offend people it's on them to get offended, I'm trying to solve a problem and think this solution will help other as well. So if we have few escape goats that get offended, so be it.

thisiskeithb, 2022-12-04:
Stop opening new issues. You will be banned.

Failing to follow of [code of conduct](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/code_of_conduct.md) will not be tolerated.