MarlinFirmware/Marlin #25056: 2004 Lcd screen not refresh - Print from Media function [BUG] (LCD Screen not Refresh)

AnandRaj-R, 2022-12-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

2004 Lcd screen not refresh, if the SD card have only one g-code file. 3nd and 4th line must be empty but it showing main menu names

### Bug Timeline

Old

### Expected behavior

If SD-card have only one g-code file then 3rd and 4th line must be empty or if SD-card have two g-code files then 4th line must be empty.
Example: if SD-card have only one g-code file
Line 1: Main
Line 2: Xyz123.gcode
Line 3: _______________ (this line must be empty)
Line 4: _______________ (this line must be empty)

Example: if SD-card have two g-code file
Line 1: Main
Line 2: Xyz123.gcode
Line 3: Abc123.gcode
Line 4: _______________ (this line must be empty)

### Actual behavior

Preheat Custome and Motion menus are showing in 3rd and 4th line

Example: if SD-card have only one g-code file
Line 1: Main
Line 2: >Xyz123.gcode
Line 3: >Preheat Custom menu showing instead of empty
Line 4: >Preheat Custom menu showing instead of empty

Example: if SD-card have two g-code file
Line 1: Main
Line 2: Xyz123.gcode
Line 3: Abc123.gcode
Line 4: Motion menu showing instead of empty

[image]

### Steps to Reproduce

Need screen refresh

### Version of Marlin Firmware

marlin 2.1

### Printer model

Anet A8

### Electronics

MKS GenL v2.1(Ramps 1.4)

### Bed Leveling

ABL 3-point

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_2.1.1.zip](https://github.com/MarlinFirmware/Marlin/files/10148802/Marlin_2.1.1.zip)

ellensp, 2022-12-05:
Minimal Config needed to trigger the issue
Stock Configs +

SDSUPPORT
REPRAP_DISCOUNT_SMART_CONTROLLER

and enable a menu under "Print from Media", I used

HOST_SHUTDOWN_MENU_ITEM
HOST_ACTION_COMMANDS

could also use ADVANCED_PAUSE_FEATURE

Without this menu Under "Print from Media", the issue does not happen.

ellensp, 2023-05-18:
This issue no longer exists in latest Marlin

closed 2023-05-18 by ellensp

AnandRaj-R, 2023-05-18:
Yes. i noted, a new "sd Refresh" function added in side the menu "Print from Media"  thanks to all .