MarlinFirmware/Marlin #25057: [BUG] (Unable to compile for Ender 5 plus using current Bugfix and example configuration)

NoviceArtisan, 2022-12-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Attempting to compile for an Ender 5 Plus with a BTT SKR V3.0 board, only changes I've made to the Marlin example config for the Ender 5 Plus is to change the Board to the one I'm using, nothing else, and it throws up this error. Beyond my ability to understand what to do here, could you help out please? :) 

Cheers

Processing STM32H743Vx_btt (platform: ststm32@~15.4.1; board: marlin_STM32H743Vx; framework: arduino)
---------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
AttributeError: 'SConsEnvironment' object has no attribute 'IsCleanTarget':
  File "C:\Users\seita\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 186:   
    env.SConscript(item, exports="env")
  File "C:\Users\seita\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\seita\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\seita\Downloads\Marlin-bugfix-2.1.x\buildroot\share\PlatformIO\scripts\configuration.py", line 229:
    if pioutil.is_pio_build():
  File "C:\Users\seita\Downloads\Marlin-bugfix-2.1.x\buildroot\share\PlatformIO\scripts\pioutil.py", line 9:
    return not env.IsIntegrationDump() and not env.IsCleanTarget()
=================================== [FAILED] Took 2.13 seconds ===================================

Environment      Status    Duration
---------------  --------  ------------
STM32H743Vx_btt  FAILED    00:00:02.135
============================== 1 failed, 0 succeeded in 00:00:02.135 ==============================
 *  The terminal process "C:\Users\seita\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'STM32H743Vx_btt'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

### Bug Timeline

New issue

### Expected behavior

I expected it to compile smoothly without issue

### Actual behavior

It didn't 

### Steps to Reproduce

d/l latest bugfix
d/l latest Ender 5 Plus config files from Marlins repo and copy them to the Marlin folder
Open VS code
Edit the configuration.h file and change the motherboard to BOARD_BTT_SKR_V3_0
Attempt to compile.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 5 Plus

### Electronics

BTT SKR V3 and stock LCD

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-12-04:
This was broken in https://github.com/MarlinFirmware/Marlin/commit/4d03c258dcd33bb0939cee38f42216e546f06d0f

Simply revert the change in [`buildroot/share/PlatformIO/scripts/pioutil.py`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/buildroot/share/PlatformIO/scripts/pioutil.py) for now:

```diff
-    return not env.IsIntegrationDump() and not env.IsCleanTarget()
+    return not env.IsIntegrationDump()
```

NoviceArtisan, 2022-12-04:
Thanks, that got it moving along a bit. Now it's throwing up a bunch more errors.

Building in release mode
Compiling .pio\build\STM32H743Vx_btt\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\STM32H743Vx_btt\FrameworkArduinoVariant\variant_MARLIN_STM32H743VX.cpp.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_adc.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_can.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_cec.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_comp.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_comp_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_cordic.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_cortex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_crc.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_crc_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_cryp.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_cryp_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dac.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dac_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dcache.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dcmi.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dcmi_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dfsdm.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dfsdm_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dma.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dma2d.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dma_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dsi.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_dts.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_eth.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_eth_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_exti.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_fdcan.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_firewall.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_flash.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_flash_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_flash_ramfunc.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_fmac.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_fmpi2c.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_fmpi2c_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_fmpsmbus.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_fmpsmbus_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_gfxmmu.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_gpio.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_gpio_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_gtzc.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_hash.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_hash_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_hcd.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_hrtim.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_hsem.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_i2c.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_i2c_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_i2s.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_i2s_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_icache.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_ipcc.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_irda.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_iwdg.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_jpeg.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_lcd.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_lptim.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_ltdc.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_ltdc_ex.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_mdf.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_mdios.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_mdma.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_mmc.c.o
Compiling .pio\build\STM32H743Vx_btt\SrcWrapper\src\HAL\stm32yyxx_hal_mmc_ex.c.o
[... cut ...]
=================================================== [FAILED] Took 140.99 seconds ===================================================

Environment      Status    Duration
---------------  --------  ------------
STM32H743Vx_btt  FAILED    00:02:20.987
============================================== 1 failed, 0 succeeded in 00:02:20.987 ============================================== 
PS C:\Users\seita\Downloads\Marlin-bugfix-2.1.x> echo "done" >"C:\Users\seita\AppData\Local\Temp\ipc"
PS C:\Users\seita\Downloads\Marlin-bugfix-2.1.x>

thisiskeithb, 2022-12-04:
The main one is `#error "SERIAL_PORT must be from 1 to 6, or -1 for Native USB."`

NoviceArtisan, 2022-12-04:
I've edited those lines, thanks. 

From what I understand of the code for the main error, it's clearly defined to -1 for native USB? Not sure where to look for where else it would be defined?

" #if WITHIN(SERIAL_PORT, 1, 6)
  #define MYSERIAL1 MSERIAL(SERIAL_PORT)
#elif !defined(USBCON)
  #error "SERIAL_PORT must be from 1 to 6."
#elif SERIAL_PORT == -1
  #define MYSERIAL1 MSerialUSB
#else
  #error "SERIAL_PORT must be from 1 to 6, or -1 for Native USB."
#endif "

NoviceArtisan, 2022-12-04:
Duh, i'm a tool. It's undefined in the Config, ignore that. I've edited and recompiling

NoviceArtisan, 2022-12-04:
Okay, so it's got a lot further now, but throwing up antother error which I thinks a typo of some sort?

Compiling .pio\build\STM32H743Vx_btt\src\src\gcode\config\M220.cpp.o
Marlin\src\gcode\calibrate\M48.cpp:285: error: unterminated argument list invoking macro "TERN_"
  285 | #endif // Z_MIN_PROBE_REPEATABILITY_TEST
      |
Marlin\src\gcode\calibrate\M48.cpp: In static member function 'static void GcodeSuite::M48()':
Marlin\src\gcode\calibrate\M48.cpp:115:3: error: 'TERN_' was not declared in this scope
  115 |   TERN_(HAS_PTC, ptc.set_enabled(parser.boolval('C', true));
      |   ^~~~~
Marlin\src\gcode\calibrate\M48.cpp:59:24: note: to match this '{'
   59 | void GcodeSuite::M48() {
      |                        ^
*** [.pio\build\STM32H743Vx_btt\src\src\gcode\calibrate\M48.cpp.o] Error 1

thisiskeithb, 2022-12-04:
[`M48.cpp`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/gcode/calibrate/M48.cpp) is also broken. See https://github.com/MarlinFirmware/Marlin/pull/25055 for the fix:

```diff
diff --git a/Marlin/src/gcode/calibrate/M48.cpp b/Marlin/src/gcode/calibrate/M48.cpp
index 825f3713f45c..bfb3b640078b 100644
--- a/Marlin/src/gcode/calibrate/M48.cpp
+++ b/Marlin/src/gcode/calibrate/M48.cpp
@@ -112,7 +112,7 @@ void GcodeSuite::M48() {
     set_bed_leveling_enabled(false);
   #endif
 
-  TERN_(HAS_PTC, ptc.set_enabled(parser.boolval('C', true));
+  TERN_(HAS_PTC, ptc.set_enabled(parser.boolval('C', true)));
 
   // Work with reasonable feedrates
   remember_feedrate_scaling_off();
```

NoviceArtisan, 2022-12-04:
Thanks, now it's throwing up an error saying the final #endif lilne of that file isn't defined by any #If before it?

 Compiling .pio\build\STM32H743Vx_btt\src\src\gcode\config\M301.cpp.o
Marlin\src\gcode\calibrate\M48.cpp:287:2: error: #endif without #if
  287 | #endif // Z_MIN_PROBE_REPEATABILITY_TEST
      |  ^~~~~
*** [.pio\build\STM32H743Vx_btt\src\src\gcode\calibrate\M48.cpp.o] Error 1

NoviceArtisan, 2022-12-04:
( i get the feeling I should just delete that line altogether as it's an incomplete function? :S)

thisiskeithb, 2022-12-04:
Use `bugfix-2.1.x` from 5c3d87a951cee94ce3d5ac500079da705a274be4 until recent typos/bugs are fixed: https://github.com/MarlinFirmware/Marlin/archive/5c3d87a951cee94ce3d5ac500079da705a274be4.zip

NoviceArtisan, 2022-12-04:
Thank you, I've transferred the config files over and started compiling from the new repo. These are the errors it's thrown up 

Compiling .pio\build\STM32H743Vx_btt\src\src\lcd\extui\dgus\DGUSScreenHandler.cpp.o
Marlin\src\inc\Warnings.cpp:731:6: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)" [-Wcpp]
  731 |     #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"
      |      ^~~~~~~
Compiling .pio\build\STM32H743Vx_btt\src\src\lcd\extui\dgus\dgus_extui.cpp.o
Compiling .pio\build\STM32H743Vx_btt\src\src\lcd\extui\dgus\origin\DGUSDisplayDef.cpp.o
Compiling .pio\build\STM32H743Vx_btt\src\src\lcd\extui\dgus\origin\DGUSScreenHandler.cpp.o
Compiling .pio\build\STM32H743Vx_btt\src\src\lcd\extui\ui_api.cpp.o
In file included from c:\bastard2\marlin\src\inc\marlinconfigpre.h:37,
                 from Marlin\src\lcd\extui\dgus\origin\DGUSDisplayDef.cpp:27:
Marlin\src\lcd\extui\dgus\origin\DGUSDisplayDef.cpp:211:16: error: 'VP_Fan2_Percentage' was not declared in this scope; did you mean 'VP_Fan1_Percentage'?
  211 |       VPHELPER(VP_Fan##N##_Percentage, &thermalManager.fan_speed[N], ScreenHandler.DGUSLCD_PercentageToUint8, ScreenHandler.DGUSLCD_SendPercentageToDisplay), \
      |                ^~~~~~
c:\bastard2\marlin\src\core\macros.h:648:24: note: in definition of macro 'EVAL1'
  648 | #define EVAL1(V...)    V
      |                        ^
c:\bastard2\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\bastard2\marlin\src\core\macros.h:646:24: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                        ^~~~~
c:\bastard2\marlin\src\core\macros.h:647:24: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\bastard2\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\bastard2\marlin\src\core\macros.h:646:30: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                              ^~~~~
c:\bastard2\marlin\src\core\macros.h:645:24: note: in expansion of macro 'EVAL4'
  645 | #define EVAL8(V...)    EVAL4(EVAL4(V))
      |                        ^~~~~
c:\bastard2\marlin\src\core\macros.h:647:24: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\bastard2\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\bastard2\marlin\src\core\macros.h:646:24: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                        ^~~~~
c:\bastard2\marlin\src\core\macros.h:647:24: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\bastard2\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\bastard2\marlin\src\core\macros.h:646:30: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                              ^~~~~
c:\bastard2\marlin\src\core\macros.h:645:30: note: in expansion of macro 'EVAL4'
  645 | #define EVAL8(V...)    EVAL4(EVAL4(V))
      |                              ^~~~~
c:\bastard2\marlin\src\core\macros.h:644:24: note: in expansion of macro 'EVAL8'
  644 | #define EVAL16(V...)   EVAL8(EVAL8(V))
      |                        ^~~~~
c:\bastard2\marlin\src\core\macros.h:647:24: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\bastard2\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\bastard2\marlin\src\core\macros.h:646:24: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                        ^~~~~
c:\bastard2\marlin\src\core\macros.h:647:24: note: in expansion of macro 'EVAL1'
[... 298 lines cut ...]
      |                    ^~~~~~~
c:\bastard2\marlin\src\core\macros.h:637:24: note: in expansion of macro 'EVAL16'
  637 | #define EVAL(V...)     EVAL16(V)
      |                        ^~~~~~
c:\bastard2\marlin\src\core\macros.h:227:22: note: in expansion of macro '_IF_1'
  227 | #define _CAT(a,V...) a##V
      |                      ^
c:\bastard2\marlin\src\core\macros.h:683:20: note: in expansion of macro '_REPEAT'
  683 | #define __REPEAT() _REPEAT
      |                    ^~~~~~~
c:\bastard2\marlin\src\core\macros.h:694:33: note: in expansion of macro 'EVAL'
  694 | #define REPEAT_S(S,N,OP)        EVAL(_REPEAT(S,SUB##S(N),OP))
      |                                 ^~~~
c:\bastard2\marlin\src\core\macros.h:227:22: note: in expansion of macro '_IF_1'
  227 | #define _CAT(a,V...) a##V
      |                      ^
c:\bastard2\marlin\src\core\macros.h:694:38: note: in expansion of macro '_REPEAT'
  694 | #define REPEAT_S(S,N,OP)        EVAL(_REPEAT(S,SUB##S(N),OP))
      |                                      ^~~~~~~
c:\bastard2\marlin\src\core\macros.h:695:33: note: in expansion of macro 'REPEAT_S'
  695 | #define REPEAT(N,OP)            REPEAT_S(0,N,OP)
Marlin\src\lcd\extui\dgus\origin\DGUSDisplayDef.cpp:214:5: note: in expansion of macro 'REPEAT'
  214 |     REPEAT(FAN_COUNT, FAN_VPHELPER)
      |     ^~~~~~
*** [.pio\build\STM32H743Vx_btt\src\src\lcd\extui\dgus\origin\DGUSDisplayDef.cpp.o] Error 1

NoviceArtisan, 2022-12-04:
I'm guessing I just need to throw  #DIAG_JUMPERS_REMOVED somewhere in the main config file. No idea about the others

thisiskeithb, 2022-12-04:
> I'm guessing I just need to throw #DIAG_JUMPERS_REMOVED somewhere in the main config file. No idea about the others

No, that's just a warning to remind users of common config/hardware issues since DIAG jumpers may be installed from teh factory.

> These are the errors it's thrown up

The stock DGUS screen may not be compatible with the SKR 3/STM32H7.

NoviceArtisan, 2022-12-04:
Is there active development on the TFT interfaces for the STM32H7 I'm more than happy to spend as long as it takes to iron out any wrinkles and test any/everything with the hardware I've got, and if so, how would I find the right people :) 

I've also got a BOARD_BTT_SKR_MINI_E3_V3_0 (STM32G0B1RE) to configure for the same machine too, which I'll attempt to build using the repo you shared. Shoud I continue on this issue for that board too, or make a new one if errors arise?

NoviceArtisan, 2022-12-04:
Bah.. it's also thrown up the same errors -Compiling .pio\build\STM32G0B1RE_btt\src\src\lcd\extui\dgus\origin\DGUSScreenHandler.cpp.o
In file included from c:\minie3v3\marlin\src\inc\marlinconfigpre.h:37,
                 from Marlin\src\lcd\extui\dgus\origin\DGUSDisplayDef.cpp:27:
Marlin\src\lcd\extui\dgus\origin\DGUSDisplayDef.cpp:211:16: error: 'VP_Fan2_Percentage' was not declared in this scope; did you mean 'VP_Fan1_Percentage'?
  211 |       VPHELPER(VP_Fan##N##_Percentage, &thermalManager.fan_speed[N], ScreenHandler.DGUSLCD_PercentageToUint8, ScreenHandler.DGUSLCD_SendPercentageToDisplay), \
      |                ^~~~~~
c:\minie3v3\marlin\src\core\macros.h:648:24: note: in definition of macro 'EVAL1'
  648 | #define EVAL1(V...)    V
      |                        ^
c:\minie3v3\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\minie3v3\marlin\src\core\macros.h:646:24: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                        ^~~~~
c:\minie3v3\marlin\src\core\macros.h:647:24: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\minie3v3\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\minie3v3\marlin\src\core\macros.h:646:30: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                              ^~~~~
c:\minie3v3\marlin\src\core\macros.h:645:24: note: in expansion of macro 'EVAL4'
  645 | #define EVAL8(V...)    EVAL4(EVAL4(V))
      |                        ^~~~~
c:\minie3v3\marlin\src\core\macros.h:647:24: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\minie3v3\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\minie3v3\marlin\src\core\macros.h:646:24: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                        ^~~~~
c:\minie3v3\marlin\src\core\macros.h:647:24: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\minie3v3\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\minie3v3\marlin\src\core\macros.h:646:30: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                              ^~~~~
c:\minie3v3\marlin\src\core\macros.h:645:30: note: in expansion of macro 'EVAL4'
  645 | #define EVAL8(V...)    EVAL4(EVAL4(V))
      |                              ^~~~~
c:\minie3v3\marlin\src\core\macros.h:644:24: note: in expansion of macro 'EVAL8'
  644 | #define EVAL16(V...)   EVAL8(EVAL8(V))
      |                        ^~~~~
c:\minie3v3\marlin\src\core\macros.h:647:24: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\minie3v3\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\minie3v3\marlin\src\core\macros.h:646:24: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                        ^~~~~
c:\minie3v3\marlin\src\core\macros.h:647:24: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                        ^~~~~
c:\minie3v3\marlin\src\core\macros.h:647:30: note: in expansion of macro 'EVAL1'
  647 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
c:\minie3v3\marlin\src\core\macros.h:646:30: note: in expansion of macro 'EVAL2'
  646 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                              ^~~~~
c:\minie3v3\marlin\src\core\macros.h:645:24: note: in expansion of macro 'EVAL4'
[... 289 lines cut ...]
      |                    ^~~~~~~
c:\minie3v3\marlin\src\core\macros.h:637:24: note: in expansion of macro 'EVAL16'
  637 | #define EVAL(V...)     EVAL16(V)
      |                        ^~~~~~
c:\minie3v3\marlin\src\core\macros.h:227:22: note: in expansion of macro '_IF_1'
  227 | #define _CAT(a,V...) a##V
      |                      ^
c:\minie3v3\marlin\src\core\macros.h:683:20: note: in expansion of macro '_REPEAT'
  683 | #define __REPEAT() _REPEAT
      |                    ^~~~~~~
c:\minie3v3\marlin\src\core\macros.h:694:33: note: in expansion of macro 'EVAL'
  694 | #define REPEAT_S(S,N,OP)        EVAL(_REPEAT(S,SUB##S(N),OP))
      |                                 ^~~~
c:\minie3v3\marlin\src\core\macros.h:227:22: note: in expansion of macro '_IF_1'
  227 | #define _CAT(a,V...) a##V
      |                      ^
c:\minie3v3\marlin\src\core\macros.h:694:38: note: in expansion of macro '_REPEAT'
  694 | #define REPEAT_S(S,N,OP)        EVAL(_REPEAT(S,SUB##S(N),OP))
      |                                      ^~~~~~~
c:\minie3v3\marlin\src\core\macros.h:695:33: note: in expansion of macro 'REPEAT_S'
  695 | #define REPEAT(N,OP)            REPEAT_S(0,N,OP)
Marlin\src\lcd\extui\dgus\origin\DGUSDisplayDef.cpp:214:5: note: in expansion of macro 'REPEAT'
  214 |     REPEAT(FAN_COUNT, FAN_VPHELPER)
      |     ^~~~~~
*** [.pio\build\STM32G0B1RE_btt\src\src\lcd\extui\dgus\origin\DGUSDisplayDef.cpp.o] Error 1

NoviceArtisan, 2022-12-04:
Okay, so I followed the suggestion in the terminals output and changed

 VPHELPER(VP_Fan##N##_Percentage, &thermalManager.fan_speed[N], 
 
 to
 
 VPHELPER(VP_Fan1_Percentage, &thermalManager.fan_speed[N], 

and it's compiled sucessfully with zero errors on both of the boards...

I assume that means that there should be an option created somewhere in the configs that ultimately fills that ##n## with the relevant fan number chosen on the board for the display to choose to interact with?

Fingers crossed for the LCD actually working as it should now hahahaha

NoviceArtisan, 2022-12-04:
(Using the Repo suppied by you from https://github.com/MarlinFirmware/Marlin/commit/5c3d87a951cee94ce3d5ac500079da705a274be4 )

NoviceArtisan, 2022-12-04:
Closing this issue as I'm now able to compile the firmware. Screen isn't giving any response other than the generic marlin screen when connected though, so i'll start a new issue for that.

closed 2022-12-04 by NoviceArtisan

thisiskeithb, 2022-12-04:
> Screen isn't giving any response other than the generic marlin screen when connected though, so i'll start a new issue for that.

No need. We already have an open issue: https://github.com/MarlinFirmware/Marlin/issues/23138