MarlinFirmware/Marlin #25059: [BUG] (DWIN DMT48270C043_06WT LCD not working with BTT SKR 3 or SKR mini e3 v3 on latest bugfix)

NoviceArtisan, 2022-12-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to get the DWIN DMT48270C043_06WT LCD to work with the BTT SKR 3 or SKR mini e3 v3 on latestish bugfix, got support on previos issue post to fix the compilation issues nicely, but sadly there's no response other than the screen displaying the Marlin logo on the LCD when connected and powered up. 

How do I troubleshoot/investigate this further so I can get it working?

### Expected behavior

I expected the LCD to work

### Actual behavior

No data transmitted of recieved (or at least none that's displayed on the LCD lol)

### Steps to Reproduce

Compile latestish bugfix with board option set for the SKR 3 or SKR mini e3 v3 and flash to board

Flash Desuuuu's  DGUS firmware on the DWIN DMT48270C043_06WT LCD 

Connect the LCD to the TFT port

Power on

Cry

### Version of Marlin Firmware

latest bugfix

### Printer model

Ender 5 Plus

### Electronics

BTT SKR V3 and stock LCD

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-12-04:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/23138

closed 2022-12-04 by thisiskeithb

NoviceArtisan, 2022-12-04:
> Duplicate of #23138

No, it's not a duplicate. DIfferent boards, different issue. Not  HiPrecy DGUS LCD and not  Fysetc S6 V2.0 controller 

Couldn't be less similar

thisiskeithb, 2022-12-04:
It's the same underlying DGUS issue. Doesn't matter what board you're using with it.

You even pasted a log that had a `VP_FAN2_CONTROL` error in https://github.com/MarlinFirmware/Marlin/issues/25057#issuecomment-1336493161.

NoviceArtisan, 2022-12-04:
Well, I see the issue with the config file, it hadn't changed to the RELOADED dgus option, so the firmware I needed to flash to the LCD wouldn't have worked anyway due to wrong driver lol

After adjusting that to the correct setting, I'm now getting this error

Compiling .pio\build\STM32H743Vx_btt\src\src\HAL\STM32\Servo.cpp.o
In file included from c:\bastard2\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
c:\bastard2\marlin\src\inc\SanityCheck.h:4251:6: error: #error "DGUS_LCD_UI_RELOADED requires LCD_BED_TRAMMING."
 4251 |     #error "DGUS_LCD_UI_RELOADED requires LCD_BED_TRAMMING."
      |      ^~~~~
In file included from c:\bastard2\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
c:\bastard2\marlin\src\inc\SanityCheck.h:4251:6: error: #error "DGUS_LCD_UI_RELOADED requires LCD_BED_TRAMMING."
 4251 |     #error "DGUS_LCD_UI_RELOADED requires LCD_BED_TRAMMING."
      |      ^~~~~
*** [.pio\build\STM32H743Vx_btt\src\src\HAL\STM32\HAL.cpp.o] Error 1
In file included from c:\bastard2\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
c:\bastard2\marlin\src\inc\SanityCheck.h:4251:6: error: #error "DGUS_LCD_UI_RELOADED requires LCD_BED_TRAMMING."
 4251 |     #error "DGUS_LCD_UI_RELOADED requires LCD_BED_TRAMMING."
      |      ^~~~~
*** [.pio\build\STM32H743Vx_btt\src\src\HAL\STM32\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32H743Vx_btt\src\src\HAL\STM32\MarlinSerial.cpp.o] Error 1
In file included from c:\bastard2\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
ED_TRAMMING."
 4251 |     #error "DGUS_LCD_UI_RELOADED requires LCD_BED_TRAMMING."
      |      ^~~~~
*** [.pio\build\STM32H743Vx_btt\src\src\HAL\STM32\Servo.cpp.o] Error 1
================================== [FAILED] Took 155.64 seconds ================================== 

Environment      Status    Duration
---------------  --------  ------------
STM32H743Vx_btt  FAILED    00:02:35.637
============================== 1 failed, 0 succeeded in 00:02:35.637 ==============================

thisiskeithb, 2022-12-04:
> error: #error "DGUS_LCD_UI_RELOADED requires LCD_BED_TRAMMING."

The error is right there in the output. Enable `LCD_BED_TRAMMING`.

NoviceArtisan, 2022-12-04:
I realised that just as soon as I pressed send. Sorry, I'm quite new to this Marlin thing, waaaay more dependencies and configurations than anythign I've ever used arduino code for before. I'll continue only when I've got stuck on a non-obvious solution lol :)

NoviceArtisan, 2022-12-04:
Hit one that's not obvious, and doesn't have a reference in the config files for me to adjust

Marlin\src\lcd\extui\dgus_reloaded\config/DGUS_Constants.h:28:37: error: static assertion failed: DGUS_LEVEL_GRID_SIZE incompatible with current mesh.
   28 | static_assert((DGUS_LEVEL_GRID_SIZE == GRID_MAX_POINTS_X * GRID_MAX_POINTS_Y), "DGUS_LEVEL_GRID_SIZE incompatible with current mesh.");
      |               ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\STM32H743Vx_btt\src\src\lcd\extui\dgus_reloaded\DGUSDisplay.cpp.o] Error 1
In file included from Marlin\src\lcd\extui\dgus_reloaded\DGUSScreenHandler.h:27,
                 from Marlin\src\lcd\extui\dgus_reloaded\DGUSRxHandler.cpp:29:
Marlin\src\lcd\extui\dgus_reloaded\config/DGUS_Constants.h:28:37: error: static assertion failed: D   28 | static_assert((DGUS_LEVEL_GRID_SIZE == GRID_MAX_POINTS_X * GRID_MAX_POINTS_Y), "DGUS_LEVEL_GRID_SIZE incompatible with current mesh.");
      |               ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\STM32H743Vx_btt\src\src\lcd\extui\dgus_reloaded\DGUSRxHandler.cpp.o] Error 1

NoviceArtisan, 2022-12-04:
Okay, so searching for each little blob of text in the error message is the way.. Discovered that the final pointer value was expecting 25 as the total, so adjusted every instance of grid_max_points_x to 5 and it's compiled :) wish me luck with this one.

NoviceArtisan, 2022-12-04:
Going to request that the ender 5 plus configuration.h be amended to include this at line 2532, not sure how best to ask for the permament addition?

#if ENABLED(DGUS_LCD_UI_RELOADED)
  #define LCD_SERIAL_PORT 1  // LCD will not work without this been set to 1
  #define SERIAL_PORT -1 // LCD will conflict with the default SERIAL_PORT 1 given further up in file otherwise
#endif

The undefined LCD_SERIAL_PORT is the root cause of that DGUS LCD not working with the BTT SKR 3 and BTT SKR MINI E3 V3 boards. 

It's now working absolutely perfectly. Thank you very much for your support and patience with getting me thru the bumps at the start

thisiskeithb, 2022-12-04:
> The undefined LCD_SERIAL_PORT is the root cause of that DGUS LCD not working with the BTT SKR 3 and BTT SKR MINI E3 V3 boards.

A new config will need to be added for different boards since this works fine on the stock board/config.

NoviceArtisan, 2022-12-04:
How do I help with that? (first time using the collab side of github tbh I'm usually leeching)

Do I need to upload the repo I've edited or just the config files or something? :) 

Cheers

thisiskeithb, 2022-12-04:
You can make a request for a an SKR 3 and SKR Mini E3 V3 configs for the Ender-5 Plus in https://github.com/MarlinFirmware/Configurations/issues/649