MarlinFirmware/Marlin #25061: [FR] (feature summary)

lunartester, 2022-12-04:
### Is your feature request related to a problem? Please describe.

Resetting X,Y,E via home with dual retraction setting.

### Are you looking for hardware support?

32-bit board supported by Marlin

### Describe the feature you want

I need someone skilled enough to produce a script + Gcode for Prusa Slicer to do the following:

    remove the feature on Marlin where it resets extruder absolute coordinate on each layer change, and simply leave it as normal Absolute for the whole print
    Allow the end user to introduce homing procedure on X,Y,E with a reset to based on ability to decide after how many printing layers the absolute coordinates will home and reset X,Y and E
    Marlin has fimrware retraction option and second extension retraction not sure if they are separate or tied up:

https://marlinfw.org/docs/features/fwretract.html
https://marlinfw.org/docs/gcode/M208.html

I'm not sure if we can trick Marlin or Prusa slicer into using the same extruder as if we are doing tool change, but I need some how to have one retraction on regular pritns and ability to have different retraction on that homing procedure, so maybe with some type of tool change retraction that second retraction can be implemented and give me the ability to use different values on each retraction.

Prusa slicer has the option to print with second extruder, maybe it's easier to trick it with a script to think we have a tool change on each homing reset procedure and we need differnt retraction on that tool change homing move vs. tricking Marlin to think we have tool change, which is essentially the same thing, I simply do not know how does Marlin behave on tool change and if it overrides anything we do in Prusa slicer or not.

Cheers

thisiskeithb, 2022-12-04:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/25054

thisiskeithb, 2022-12-04:
Stop opening new issues. You will be banned.

closed 2022-12-04 by thisiskeithb

lunartester, 2022-12-04:
stop closing, its not for you to decide when it's resolved or not

thisiskeithb, 2022-12-04:
> stop closing, its not for you to decide when it's resolved or not

It is 100% up to me and others within the Marlin project. You're behavior and actions are toxic and will not be tolerated.

lunartester, 2022-12-04:
fuck you then, you need to have your ego checked at the door when you moderate on who can participate and who can't